JenniferBuehler/gazebo-pkgs #2: Grasp plugin for multi-gripper robot

mexomagno, 2016-03-19:
Hi again, Jenny.
I've been testing some stuff and as I really didn't need to use more than one gripper, your plugin was working wonderfully :).
However, when I tried to use the other gripper, there were only problems.

In the robot urdf description (technically xacro) I tried the following configurations:
- First try
```
<gazebo>
    <plugin name=...>
        <palm_link>left_gripper_palm_link</palm_link>
        <palm_link>right_gripper_palm_link</palm_link>
        <!-- gripper links from both grippers-->
        ...
    </plugin>
</gazebo>
```
This results in your plugin nicely detecting grips from both grippers, but it attaches the object to the first palm link specified. So in this case, if right gripper grabs a cube, the cube gets attached to the left gripper and weird stuff happens.

- Second try
```
<gazebo>
    <plugin name="leftgripperfix" ...>
        <!-- palm and gripper links for LEFT gripper -->
    </plugin>
    <plugin name="rightgripperfix"...>
        <!-- palm and gripper links for RIGHT gripper -->
   </plugin>
</gazebo>
```
This results on the plugin working only on the left gripper. The right one doesn't, I think it's because "`Filter with the same name already exists! Aborting`".

- Third and last try
```
<gazebo>
    <plugin name="leftgripperfix"...>
        <!-- left gripper -->
    </plugin>
</gazebo>
<gazebo>
    <plugin name="rightgripperfix"...>
        <!-- right gripper -->
    </plugin>
</gazebo>
```
Same as the previous case. Same error message.

Is there a way to make it work with both grippers?
Cheers!

JenniferBuehler, 2016-03-22:
Hi Max,

yes, this is currently a limitation of the plugin. I've described this in the class header.
At the moment only one object can be grasped at a time, and only one arm can be used per robot. It should not be too hard to change this though. I will hopefully be able to address this as soon as possible, maybe in the next week or two.

I was going to try what you did in your second (and third, which has the same effect) try as well, but I was already suspecting that it won't work.

Your first try doesn't work because only one palm_link will be loaded.

Meanwhile, if you urgently need it for both arms, you could make a wrapper for the plugin which loads different arms in the Load() method and initializes the plugin with the according details. Or you can wait until I've done it.

I will post here as soon as I've removed this limitation.

Cheers
Jenny

mexomagno, 2016-03-22:
Thanks for the clarification. I didn't see that in the header.

It's not terribly urgent for my current case use as I'm really grasping with only one arm at a time. When I use the other gripper I'll have to just edit the plugin links and restart the simulation. Not beautiful, but doable and perfectly functional :)

Cheers Jenny.

JenniferBuehler, 2016-03-25:
Hi Max,

I am back to working now and will do the updates to the plugins in the next few days. Will report in this issue as soon as it's done.

P.S. I'm near Valparaiso right now actually, viva Chile!

JenniferBuehler, 2016-03-29:
Right, so I've done some changes and now it *should* be possible to use several arms. However in my current test-bed I don't have a robot with several arms and it will probably take me longer to set it up than it will take you to test it quickly. Can you please help me test it?
Please note there are now new description definitions to [the graspit plugin](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin), you have to put the palm and fingers into an ``<arm>`` tag.
It works for a single arm as it did before, but it's possible that there are a few minor mistakes for grasping with two different grippers. Please let me know if it works. Sorry for being too lazy to quickly set up a new test-bed, I have lots of other things on my plate right now ;)

Cheers
Jenny

mexomagno, 2016-04-08:
Hi!
I haven´t been able to test your changes, as I'm not currently working on this. I'll try to do some tests soon so you can confirm it works, but I'm quite busy.
Just so you know, a friend started integrating his grasp detection algorithm in a similar environment as mine, so he probably will be using your awesome and magic plugin of wonders.
Sorry for the delay.
Max

P.S.: I live in Santiago! If you are still here and happen to be near, beers are definitely on me.

hvpandya, 2017-08-02:
I have been using the plugin for a few projects with multiple arms/grippers and it works just fine by adding new <arm> tags for individual arms.

JenniferBuehler, 2017-08-04:
Awesome, thanks for letting me know @hvpandya :))

closed 2017-08-04 by JenniferBuehler