JenniferBuehler/gazebo-pkgs #3: Spawning a cylinder instead of a cube

ditoec, 2016-03-24:
Hi Jennifer,

I tried your cube spawner and it works beautifully. Since I need to simulate cylinder object as well, so I tried to modify your cube spawner to be a cylinder spawner. I assume that I need to modify cube_spawner.cpp and cube_spawner_node.cpp, is that right? So I have made some adjustments for sdf file generator such as inertia calculation and geometry & collision dimension declaration. The code compiled without warning, and I am able to spawn the cylinder to gazebo. The problem is, I cannot see the cylinder in gazebo visualization, even though it is detected on the gazebo object list. Do you have an idea where's the problem came from? If this doesn't make sense to you then I will post the code, Thanks.

JenniferBuehler, 2016-03-25:
Yes you would need to modify the SDF code which is generated by cube_spawner.cpp. Somewhat ugly, it's hard-coded. Just didn't bother to change it as it is only a test node.
Can you please post the modifications to the code you've made? That will help me to see where the problem could be.
The inertia and collision definitions should not matter. The visual / geometry definitions are probably the issue.

Cheers

JenniferBuehler, 2016-03-25:
Yes you would need to modify the SDF description which is generated by cube_spawner.cpp. Somewhat ugly, it's hard-coded. Just didn't bother to change it as it is only a test node.
Can you please post the modifications to the code you've made? That will help me to see where the problem could be.
The inertia and collision definitions should not matter. The visual / geometry descriptions are probably the issue.

Cheers

ditoec, 2016-03-25:
OK so here is the code for cube_spawner ::

`
#include <gazebo_test_tools/gazebo_cylinder_spawner.h>

#include <sstream>

using gazebo_test_tools::GazeboCylinderSpawner;
using ros::NodeHandle;

#define SPAWN_OBJECT_TOPIC "gazebo/spawn_sdf_model"

GazeboCylinderSpawner::GazeboCylinderSpawner(NodeHandle &n) : nh(n){
    spawn_object = n.serviceClient<gazebo_msgs::SpawnModel>(SPAWN_OBJECT_TOPIC);
}

void GazeboCylinderSpawner::spawnCylinder(const std::string& name, const std::string& frame_id, float x, float y, float z, float qx, float qy, float qz, float qw, float height, float radius, float _mass) {

    geometry_msgs::Pose pose;
    pose.position.x=x;
    pose.position.y=y;
    pose.position.z=z;
    pose.orientation.x=qx;
    pose.orientation.y=qy;
    pose.orientation.z=qz;
    pose.orientation.w=qw;

    gazebo_msgs::SpawnModel spawn;
    spawn.request.model_name=name;

    float h=height;
    float r=radius;

    float mass=_mass;
    float mass12=mass/12.0;

    double mu1=500; //500 for PR2 finger tip. In first experiment had it on 1000000
    double mu2=mu1;
    double kp=10000000; //10000000 for PR2 finger tip
    double kd=1; //100 for rubber? 1 fir OR2 finger tip

    bool do_surface=false;
    bool do_inertia=true;

    std::stringstream s;\
    s<<"<?xml version='1.0'?>\
    <sdf version='1.4'>\
    <model name='"<<name<<"'>\
        <static>false</static>\
        <link name='link'>";
    if (do_inertia) 
        s<<"<inertial>\
        <mass>"<<mass<<"</mass>\
        <inertia>\
          <ixx>"<<mass12*(3*r*r+h*h)<<"</ixx>\
          <ixy>0.0</ixy>\
          <ixz>0.0</ixz>\
          <iyy>"<<mass12*(3*r*r+h*h)<<"</iyy>\
          <iyz>0.0</iyz>\
          <izz>"<<mass12*(6*r*r)<<"</izz>\
        </inertia>\
          </inertial>";
         
    s<<"<collision name='collision'>\
        <geometry>\
          <cylinder>\
            <length>"<<h<<"</lenght>\
	    <radius>"<<r<<"</radius>\
          </cylinder>\
        </geometry>";
    if (do_surface)
        s<<"<surface>\
            <friction>\
              <ode>\
            <mu>"<<mu1<<"</mu>\
            <mu2>"<<mu2<<"</mu2>\
            <fdir1>0.000000 0.000000 0.000000</fdir1>\
            <slip1>0.000000</slip1>\
            <slip2>0.000000</slip2>\
              </ode>\
            </friction>\
            <bounce>\
              <restitution_coefficient>0.000000</restitution_coefficient>\
              <threshold>100000.000000</threshold>\
            </bounce>\
            <contact>\
              <ode>\
            <soft_cfm>0.000000</soft_cfm>\
            <soft_erp>0.200000</soft_erp>\
            <kp>"<<kp<<"</kp>\
            <kd>"<<kd<<"</kd>\
            <max_vel>100.000000</max_vel>\
            <min_depth>0.001000</min_depth>\
              </ode>\
            </contact>\
        </surface>";
      s<<"</collision>\
          <visual name='visual'>\
        <geometry>\
          <cylinder>\
            <length>"<<h<<"</lenght>\
	    <radius>"<<r<<"</radius>\
          </cylinder>\
        </geometry>\
        <material>\
            <script>\
                <uri>file://media/materials/scripts/gazebo.material</uri> \
                <name>Gazebo/Blue</name>\
            </script>\
        </material>\
          </visual>\
        </link>\
      </model>\
    </sdf>";
    
    spawn.request.model_xml=s.str();
    spawn.request.robot_namespace="cylinder_spawner";
    spawn.request.initial_pose=pose;
    spawn.request.reference_frame=frame_id;

    ROS_INFO("Resulting model: \n %s",s.str().c_str());

    //ROS_INFO("Waiting for service");
    spawn_object.waitForExistence();
    //ROS_INFO("Calling service");

    //std::cout<<spawn.request<<std::endl;

    if (!spawn_object.call(spawn)) {
        ROS_ERROR("Failed to call service %s",SPAWN_OBJECT_TOPIC);
    }
    ROS_INFO("Result: %s, code %u",spawn.response.status_message.c_str(), spawn.response.success);
}
`

I call the function like this ::

`dito@dito-PC:~$ rosrun gazebo_test_tools cylinder_spawner cylinder1 0.35 0 1`
`[ INFO] [1458901455.165514360]: Running spawn cylinder once..`
`[ INFO] [1458901455.312876524]: Result: SpawnModel: Successfully spawned model, code 1`

And I run request object info like this ::

> dito@dito-PC:~$ rosrun gazebo_state_plugins gazebo_request_object_info cylinder1
[ INFO] [1458901582.195193173]: Got objects topic name: </gazebo_objects/get_info>
[ INFO] [1458901582.200430890]: Result:
success: 1
object: 
  header: 
    seq: 0
    stamp: 4441.578000000
    frame_id: world
  name: cylinder1
  type: 
    key: 
    db: 
  primitives[]
    primitives[0]: 
        type: 1
      dimensions[]
        dimensions[0]: 2.21384
        dimensions[1]: 2.21384
        dimensions[2]: 2.21384
  primitive_poses[]
    primitive_poses[0]: 
        position: 
        x: -0.228194
        y: 0.202584
        z: 0.688075
      orientation: 
        x: 0.135374
        y: -0.31899
        z: -0.366447
        w: 0.863502
  meshes[]
  mesh_poses[]
  planes[]
  plane_poses[]
  primitive_origin: -1
  mesh_origin: -2
  origin: 
    position: 
      x: -0.228194
      y: 0.202584
      z: 0.688075
    orientation: 
      x: 0.135374
      y: -0.31899
      z: -0.366447
      w: 0.863502
  content: 0
error_code: 0

But still, the cylinder itself is nowhere to be seen inside Gazebo, clicking on the 'cylinder1' object in models list also doesn't give me the bounding box.

Thanks for your help.

JenniferBuehler, 2016-03-28:
Sorry, I only got around to check on this now.
You have a typo in the geometry specification, it should be ``</length>``, not ``</lenght>``.
It works for me if I change only the following lines at the two occurrences:

Swap the box:
```
          <box>\
            <size>"<<w<<" "<<h<<" "<<d<<"</size>\
          </box>\
```

for the cylinder:

```
            <cylinder>\
                <length>"<<h<<"</length>\
                <radius>"<<w<<"</radius>\
            </cylinder>\
```
This uses the width as radius so not nice, but the test showed it worked.

As said, not pretty that this is hard-coded, but it's only a test method. I have changed the function slightly so that it's possible to swap between cube and cylinder (also computing according interia). There is now a function spawnPrimitive(). Let me know if it works ok, I haven't tested the interia properties, but according to [this](https://en.wikipedia.org/wiki/List_of_moments_of_inertia) it should be correct.

closed 2016-03-28 by JenniferBuehler