JenniferBuehler/gazebo-pkgs #4: Magnetic gripper physics not working

harshmdeshpande, 2016-07-27:
I am trying to simulate a magnetic gripper in Gazebo which has **no moving parts**. I am successfully able to [visualize the inertia](http://i.imgur.com/lg4UTq8.png). 

There are some scenarios in my simulation where the gripper has picked up the tool (wrench) and uses it to rotate a valve. For now, to simplify the problem I have attached the tool to the gripper directly in URDF and using it to manipulate the objects. 

But I see that the robot and object vibrate frantically and fly off when in contact. https://youtu.be/eP29QGYoQhY. 

I was trying to set the physics right and then came across your plugin. But I realize that your plugin needs that the gripper should have 2 fingers which exert opposing forces on the object. As I mentioned  earlier, my gripper has no moving parts. Also I don't need to attach the valve to the tool but need it to remain stable during the simulation. 

Can I still use your plugin or can I modify it in some way?
[gripper.tar.gz](https://github.com/JenniferBuehler/gazebo-pkgs/files/385612/gripper.tar.gz)

JenniferBuehler, 2016-07-27:
Hello,

Yes the frantic vibration is an issue with the Gazebo physics engine(s). That's the reason I wrote the plugin. It works by attaching the object to the robot model, so the physics engine doesn't mess things up (because the object is considered part of the robot).
The same would need to be done with your valve then - you'd have to attach it to the robot. I'm not sure if this would work if the valve is attached to a joint in another model, this could still lead to frantic vibrations as the robot turns the valve, but it's worth testing it.

You'd have to rewrite the plugin, but for a first quick test I would write only a little plugin which attaches any object the magnetic gripper touches. You could use my plugin as a template and remove all the two-finger related stuff. The condition to "attach" is then simply that one contact point exists (instead of the complicated evaluation of contact points).
If with this simple test the robot doesn't explode or do any weird things any more, then it's worth investigating further to do a similar plugin which is a bit more elaborate than the 1-contact condition (and which also includes release conditions etc).

Does this make sense? Let me know if you need any help with how to approach this.

harshmdeshpande, 2016-07-29:
OK I think I finally understood what you have been telling me. I will try this solution and get back to you.
Thanks a ton! Will still keep the issue open in case I need your help again.

JenniferBuehler, 2016-11-27:
Closing this because of inactivity and because it's not a specific issue related to the code which needs to be resolved. Feel free to re-open if you continue working on this.

closed 2016-11-27 by JenniferBuehler

harshmdeshpande, 2016-11-28:
Sure Jennifer