JenniferBuehler/gazebo-pkgs #6: Error catkin_make

nikkk91, 2016-11-23:
I try to compile gazebo-pkgs after installing all the other packages for ros kinetic in Ubuntu 16.04 but i receive these errors. What can i do?

from /home/username/catkin_ws/src/gazebo-pkgs/gazebo_grasp_plugin/src/GazeboGraspGripper.cpp:2:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /usr/include/c++/5/random:35:0,
                 from /usr/include/ignition/math2/ignition/math/Rand.hh:20,
                 from /usr/include/ignition/math2/ignition/math.hh:18,
                 from /usr/include/sdformat-4.0/sdf/Param.hh:34,
                 from /usr/include/sdformat-4.0/sdf/Element.hh:24,
                 from /usr/include/sdformat-4.0/sdf/sdf.hh:5,
                 from /usr/include/gazebo-7/gazebo/common/Battery.hh:25,
                 from /usr/include/gazebo-7/gazebo/common/common.hh:8,
                 from /usr/include/gazebo-7/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/username/catkin_ws/src/gazebo-pkgs/gazebo_grasp_plugin/src/GazeboGraspFix.cpp:2:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21:0,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/username/catkin_ws/src/gazebo-pkgs/gazebo_grasp_plugin/src/GazeboGraspGripper.cpp:2:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
In file included from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21:0,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/username/catkin_ws/src/gazebo-pkgs/gazebo_grasp_plugin/src/GazeboGraspFix.cpp:2:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
gazebo-pkgs/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/build.make:86: recipe for target 'gazebo-pkgs/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspGripper.cpp.o' failed
make[2]: *** [gazebo-pkgs/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspGripper.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
gazebo-pkgs/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/build.make:62: recipe for target 'gazebo-pkgs/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspFix.cpp.o' failed
make[2]: *** [gazebo-pkgs/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspFix.cpp.o] Error 1
CMakeFiles/Makefile2:8155: recipe for target 'gazebo-pkgs/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/all' failed
make[1]: *** [gazebo-pkgs/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Scanning dependencies of target gazebo_test_tools_generate_messages_cpp
[ 87%] Generating C++ code from gazebo_test_tools/RecognizeGazeboObject.srv
[ 87%] Built target gazebo_test_tools_generate_messages_cpp
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

What can i do?

JenniferBuehler, 2016-11-23:
Hi,

the packages have only been tested with ROS Indigo and Jade, I was going to adjust them for Kinetic but I doubt I'll get around to it before Christmas, as I'm crazy busy at the moment.

The above compiler error means you need to add -std=c++11 to your compiler flags, you can do this with

``add_definitions(-std=c++11)``

in the CMakeLists.txt.
There may be other errors following in Kinetic though, as I said,  I haven't gotten around to test it yet...

Jennifer

nikkk91, 2016-11-24:
Thank you for your answer, I fixed it.
Now i have a problem with the joint_control_pkgs that returns me this error. I think that there's a problem related with some libraries of Gazebo 7. 

[ 86%] Building CXX object joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_control.dir/src/JointController.cpp.o
[ 86%] Building CXX object joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_state_publisher.dir/src/GazeboJointStatePublisher.cpp.o
In file included from /usr/include/gazebo-7/gazebo/transport/TopicManager.hh:30:0,
                 from /usr/include/gazebo-7/gazebo/transport/Node.hh:30,
                 from /home/username/catkin_ws/src/joint-control-pkgs/gazebo_joint_control/include/gazebo_joint_control/JointController.h:31,
                 from /home/username/catkin_ws/src/joint-control-pkgs/gazebo_joint_control/src/JointController.cpp:23:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_control.dir/build.make:62: recipe for target 'joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_control.dir/src/JointController.cpp.o' failed
make[2]: *** [joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_control.dir/src/JointController.cpp.o] Error 1
CMakeFiles/Makefile2:8101: recipe for target 'joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_control.dir/all' failed
make[1]: *** [joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_control.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/include/gazebo-7/gazebo/physics/Entity.hh:30:0,
                 from /usr/include/gazebo-7/gazebo/physics/Model.hh:34,
                 from /usr/include/gazebo-7/gazebo/physics/Actor.hh:24,
                 from /usr/include/gazebo-7/gazebo/physics/physics.hh:2,
                 from /home/username/catkin_ws/src/joint-control-pkgs/gazebo_joint_control/include/gazebo_joint_control/GazeboJointStatePublisher.h:27,
                 from /home/username/catkin_ws/src/joint-control-pkgs/gazebo_joint_control/src/GazeboJointStatePublisher.cpp:22:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_state_publisher.dir/build.make:62: recipe for target 'joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_state_publisher.dir/src/GazeboJointStatePublisher.cpp.o' failed
make[2]: *** [joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_state_publisher.dir/src/GazeboJointStatePublisher.cpp.o] Error 1
CMakeFiles/Makefile2:8564: recipe for target 'joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_state_publisher.dir/all' failed
make[1]: *** [joint-control-pkgs/gazebo_joint_control/CMakeFiles/gazebo_joint_state_publisher.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

JenniferBuehler, 2016-11-27:
When you do a

``locate Inertial.hh``

what does it say. I guess you didn't compile Gazebo 7 from source?
If you haven't, try to install the [ignition math development packages](http://ignitionrobotics.org/libraries/math), though you should already have them... see what locate says.

Jenny

csal3, 2017-05-29:
Hi,

I had the same error as nikki91 and fixed it with add_definitions(-std=c++11). However now I am getting the following error:

[ 91%] Built target fake_object_recognizer_node
In file included from /usr/include/boost/bind.hpp:22:0,
                 from /opt/ros/kinetic/include/ros/publisher.h:35,
                 from /opt/ros/kinetic/include/ros/node_handle.h:32,
                 from /opt/ros/kinetic/include/ros/ros.h:45,
                 from /home/username/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/include/joint_trajectory_execution/TrajectoryActionServer.h:27,
                 from /home/username/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/src/TrajectoryActionServer.cpp:19:
/usr/include/boost/bind/bind.hpp: In instantiation of ‘void boost::_bi::list2<A1, A2>::operator()(boost::_bi::type<void>, F&, A&, int) [with F = boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>; A = boost::_bi::list1<const actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>; A1 = boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>; A2 = boost::arg<1>]’:
/usr/include/boost/bind/bind.hpp:905:50:   required from ‘boost::_bi::bind_t<R, F, L>::result_type boost::_bi::bind_t<R, F, L>::operator()(A1&&) [with A1 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >; R = void; F = boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>; L = boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> >; boost::_bi::bind_t<R, F, L>::result_type = void]’
/usr/include/boost/function/function_template.hpp:159:11:   required from ‘static void boost::detail::function::void_function_obj_invoker1<FunctionObj, R, T0>::invoke(boost::detail::function::function_buffer&, T0) [with FunctionObj = boost::_bi::bind_t<void, boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>, boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> > >; R = void; T0 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >]’
/usr/include/boost/function/function_template.hpp:940:38:   required from ‘void boost::function1<R, T1>::assign_to(Functor) [with Functor = boost::_bi::bind_t<void, boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>, boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> > >; R = void; T0 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >]’
/usr/include/boost/function/function_template.hpp:728:7:   required from ‘boost::function1<R, T1>::function1(Functor, typename boost::enable_if_c<boost::type_traits::ice_not<boost::is_integral<Functor>::value>::value, int>::type) [with Functor = boost::_bi::bind_t<void, boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>, boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> > >; R = void; T0 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >; typename boost::enable_if_c<boost::type_traits::ice_not<boost::is_integral<Functor>::value>::value, int>::type = int]’
/usr/include/boost/function/function_template.hpp:1077:16:   required from ‘boost::function<R(T0)>::function(Functor, typename boost::enable_if_c<boost::type_traits::ice_not<boost::is_integral<Functor>::value>::value, int>::type) [with Functor = boost::_bi::bind_t<void, boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>, boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> > >; R = void; T0 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >; typename boost::enable_if_c<boost::type_traits::ice_not<boost::is_integral<Functor>::value>::value, int>::type = int]’
/home/username/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/src/TrajectoryActionServer.cpp:270:46:   required from here
/usr/include/boost/bind/bind.hpp:313:35: error: no match for call to ‘(boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>) (joint_trajectory_execution::TrajectoryActionServer*&, const actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&)’
         unwrapper<F>::unwrap(f, 0)(a[base_type::a1_], a[base_type::a2_]);

I am using ROS kinetic and Gazebo 7. What can I do?

Thanks!

JenniferBuehler, 2017-05-30:
Hi,

did you pull and compile the version of joint-control-pkgs in the last 2 weeks, or do you have an older version?

Cheers
Jenny

csal3, 2017-05-30:
Hi, 

Thanks for the reply! I removed the packages and cloned them again and it compiled now.

Thanks!

JenniferBuehler, 2017-05-30:
Great :)

closed 2017-05-30 by JenniferBuehler