JenniferBuehler/gazebo-pkgs #7: Failed to call service /gazebo_objects/get_info

odysseusPolytropos, 2016-11-24:
Hello,

I'm trying to tinker around with the fake_object_recognition and tried to execute [the tutorial ](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/Object-information-and-recognition-tutorial). Until step 3 (spawning the cube) everything works perfectly. At step 4 I get this error:

rosuser@computer ~/test_jaco $ rosrun gazebo_state_plugins gazebo_request_object_info cube1
[ INFO] [1479977436.038761419]: Got objects topic name: </gazebo_objects/get_info>
[ERROR] [1479977436.039658131]: Failed to call service /gazebo_objects/get_info, success flag: 0

Actually the service is not running:
rosuser@computer ~/test_jaco $ rosservice list
/fake_object_recognizer/get_loggers
/fake_object_recognizer/set_logger_level
/gazebo/apply_body_wrench
/gazebo/apply_joint_effort
/gazebo/clear_body_wrenches
/gazebo/clear_joint_forces
/gazebo/delete_model
/gazebo/get_joint_properties
/gazebo/get_link_properties
/gazebo/get_link_state
/gazebo/get_loggers
/gazebo/get_model_properties
/gazebo/get_model_state
/gazebo/get_physics_properties
/gazebo/get_world_properties
/gazebo/pause_physics
/gazebo/reset_simulation
/gazebo/reset_world
/gazebo/set_joint_properties
/gazebo/set_link_properties
/gazebo/set_link_state
/gazebo/set_logger_level
/gazebo/set_model_configuration
/gazebo/set_model_state
/gazebo/set_parameters
/gazebo/set_physics_properties
/gazebo/spawn_gazebo_model
/gazebo/spawn_sdf_model
/gazebo/spawn_urdf_model
/gazebo/unpause_physics
/gazebo_test_tools/recognize_object
/object_tf_broadcaster/get_loggers
/object_tf_broadcaster/register_object
/object_tf_broadcaster/set_logger_level
/robot_state_publisher/get_loggers
/robot_state_publisher/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
/rqt_gui_py_node_7161/get_loggers
/rqt_gui_py_node_7161/set_logger_level

I'm running ROS Jade with Gazebo 5.3.0 on Mint based on Ubuntu 14.04, I checked out convenience-pkgs, gazebo-pkgs, gazebo-pkgs, jaco-arm-pkgs, joint-control-pkgs and the dependencies from the wiki. Compilation works fine and I'm able to start the gazebo simulation with the robot. Therefore I assume that something is wrong with loading the world plugin or the later loaded plugins that offer the get_info service...

Since I'm quite new to Gazebo and it's plugin system a couldn't figure out until now, where the chain is broken. Any suggestions?

JenniferBuehler, 2016-11-24:
Hello,

Just a quick acknowledge that I saw this and going to look into it, I can't make it before tomorrow though. It will probably be something gone wrong in Step... could you send the output of all steps please?

Jenny

odysseusPolytropos, 2016-11-26:
Hello Jenny,

please find the output of step 1-3 attached (I don't know how to persuade markdown not to parse the text):

[Step 1](https://github.com/JenniferBuehler/gazebo-pkgs/files/614224/step1.txt)
[Step 2](https://github.com/JenniferBuehler/gazebo-pkgs/files/614223/step2.txt)
[Step 3](https://github.com/JenniferBuehler/gazebo-pkgs/files/614225/step3.txt)

Thank you for looking into it!

JenniferBuehler, 2016-11-27:
Hello,

The service is obviously not up and running, as you pointed out. 

Can you send a ``rosparam list`` after Step 1 and both a ``rosparam list`` and a ``rosservice list`` after Step 2?
Make sure both steps are started in separate terminals (which I guess you did) and are not aborted.

Do you have a ``roscore`` running in a separate terminal in the background somewhere?

Jenny

odysseusPolytropos, 2016-11-27:
Hallo Jenny,

thank you for your reply. Here is the output of the commands:
[`rosparam list` after step 1](https://github.com/JenniferBuehler/gazebo-pkgs/files/614791/ros_param_step1.txt)
[`rosparam list` after step 2](https://github.com/JenniferBuehler/gazebo-pkgs/files/614790/ros_param_step2.txt)
[`rossercive list` after step2](https://github.com/JenniferBuehler/gazebo-pkgs/files/614789/ros_service_step2.txt)

I execute every step in a new terminal and they keep running. I tried both, running a fresh `roscore` before step 1 (after executing `killall roscore`) and executing the steps without a previously running `roscore`. I did not find any differences in terms of the erroneous behaviour.

JenniferBuehler, 2016-11-29:
Hi,

sorry for taking so long to reply. I've been crazy busy in the last days. I've had to set up a test system again as I haven't been using this code lately. I could reproduce your error, a new mistake must have snug into the code some time in one of the last pushes: The default of ``load_world_plugins`` in ``jaco_on_table/jaco_on_table_gazebo_controlled.launch`` was set to false :( I don't know why I did that, I must have been testing something and then forgotten to set it back to true.

I've pushed the update, can you try it again? 

Jenny

odysseusPolytropos, 2016-11-29:
Hello Jenny,
thank you very much for investigating. This solves my problem. Anyway I'm not able to see any commit on [`jaco_on_table_gazebo_controlled.launch`](https://github.com/JenniferBuehler/jaco-arm-pkgs/blob/master/jaco_tutorial/jaco_on_table/launch/jaco_on_table_gazebo_controlled.launch). Are you sure you pushed the commit?
Thank you once again,
Odysseus

JenniferBuehler, 2016-11-29:
Hi,

Huh, I thought I did! Will check when I get home. You can change it
yourself meanwhile, just set the default to true :)

Jenny

closed 2017-04-06 by JenniferBuehler

Abdo1994, 2018-04-03:
Hi.
I am working on a Universal Robot but I get the same error at the same step
 [ERROR] [1479977436.039658131]: Failed to call service /gazebo_objects/get_info, success flag: 0

Where should I add load_world_plugins 

I am still new to ROS

Thank you

JenniferBuehler, 2018-04-05:
From some launch file, you must launch gazebo, like this

```xml
  #####################################                                         
  # Load gazebo                                                                 
  #####################################                                         
  <include file="$(find jaco_description)/launch/jaco_gazebo.launch">           
    <arg name="robot_name" value="$(arg robot_name)"/>                          
    <arg name="x" value="$(arg x)"/>                                            
    <arg name="y" value="$(arg y)"/>                                            
    <arg name="z" value="$(arg z)"/>                                            
    <arg name="gazebo_world_file" value="$(arg gazebo_world_file)"/>            
    <arg name="load_joint_state_publisher" value="$(arg load_joint_state_publisher)"/>
    <arg name="joint_state_publisher_topic" value="$(arg joint_state_publisher_topic)"/>
    <arg name="load_robot_state_publisher" value="$(arg load_robot_state_publisher)"/>
    <arg name="joint_states_topic" value="$(arg joint_states_topic)"/>          
    <arg name="extra_gazebo_args" value="-s libgazebo_world_plugin_loader.so"/>
  </include>               
```

You must have the extra line ``extra_gazebo_args``

padhupradheep, 2019-04-02:
Hey Jenny,

I'm also using a universal robot, if I add the line with extra_gazebo_args, 

An error appears, as follows;

 [ERROR] [1554207521.577053519]: Parameter world_plugins should be specified as an array. Gazebo world plugins won't be loaded.
 
What should I be doing?