JenniferBuehler/gazebo-pkgs #10: Object moves away

csal3, 2017-06-06:
Hi,

I tried the plugin and sometimes it works perfectly. However, sometimes, as I approach the object and start closing the gripper, the object moves/flies away. Is this due to the inertia parameters of the gripper or of the object? Or do I have to change the parameters of the grasp plugin?

Thanks!

JenniferBuehler, 2017-06-08:
Hello,

that would probably happen if the object is touched before the plugin takes effect. That most likely means that the configuration of the gripper and the object are such that the physics goes a bit crazy... this is why I  developed the plugin, to avoid the robot exploding as it touches the object, and weird things like this which also happened to me (note though that I was still on Gazebo 2 when I developed this, it should have improved by now).
If you want to decrease the likelihood of this happening, you will need to tune the plugin such that it takes effect sooner, though it may not make a difference at all. You could try increase the parameter ``forces_angle_tolerance`` which would relax the condition of when the object is considered "grasped". You may also want to try ``grip_count_threshold`` and ``max_grip_count``, the parameters are explained [on the wiki](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin).

Let me know how it goes
Jenny

csal3, 2017-06-21:
Hi,

I tried changing parameters but did not see any changes with some objects. I am posting a video trying to grasp a handle, and as you can see when approaching the handle, the object moves. I tried increasing the inertia and mass of the object and seemed to behave better but at the end it still moved away.

https://www.youtube.com/watch?v=19E9M3edt_c&feature=youtu.be

JenniferBuehler, 2017-06-26:
This looks like the collision shapes don't match the visual shapes in the URDF. Can you switch viewing the collisions on in Gazebo? (View -> Collisions)
The collision shapes are what will count, not the shapes you actually see (unless they happen to be the same). The collision shapes are displayed in orange once you switch on the visualisation.

csal3, 2017-07-27:
The collision shapes of the gripper and the object are exactly the same because I am using the same mesh file with the same position. As you might have seen, the gripper is made of many links and joints. Do you think the problem might be in the inertia values of each link of the gripper? Or maybe because of the PID controller of the gripper? I am using ros control.

Thanks!

JenniferBuehler, 2017-08-12:
Hm, that doesn't make sense... in the video, the problem happens before the gripper actually touches the object. Both the gazebo collisions, as also the plugin, will only kick in once there are actual contact points. Contact points happen with the collision shapes.

The inertia would not have anything to do with this. That also only has effect once the contact happens. 
The PID controller also is not the problem, a issue with it would only lead to jittering behaviour or lack of control of the arm (e.g. collapsing).

What about the collision shape of the box? Is that the same as the visual too?

csal3, 2017-09-20:
Yes, the visual and collision shapes are the same. The problem does not seem to be arising with regard to all objects. Anyways thanks a lot for your help!!

closed 2017-09-20 by csal3