JenniferBuehler/gazebo-pkgs #13: catkin_make error.

wonwon0, 2018-01-23:
Sorry for the long error message, I am pretty puzzled by this error message. When I run catkin_make on a fresh ubuntu 16.04 install I get this:

In file included from /usr/include/boost/bind.hpp:22:0,
                 from /opt/ros/kinetic/include/ros/publisher.h:35,
                 from /opt/ros/kinetic/include/ros/node_handle.h:32,
                 from /opt/ros/kinetic/include/ros/ros.h:45,
                 from /home/phil/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/include/joint_trajectory_execution/TrajectoryActionServer.h:27,
                 from /home/phil/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/src/TrajectoryActionServer.cpp:19:
/usr/include/boost/bind/bind.hpp: In instantiation of ‘void boost::_bi::list2<A1, A2>::operator()(boost::_bi::type<void>, F&, A&, int) [with F = boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>; A = boost::_bi::list1<const actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>; A1 = boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>; A2 = boost::arg<1>]’:
/usr/include/boost/bind/bind.hpp:905:50:   required from ‘boost::_bi::bind_t<R, F, L>::result_type boost::_bi::bind_t<R, F, L>::operator()(A1&&) [with A1 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >; R = void; F = boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>; L = boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> >; boost::_bi::bind_t<R, F, L>::result_type = void]’
/usr/include/boost/function/function_template.hpp:159:11:   required from ‘static void boost::detail::function::void_function_obj_invoker1<FunctionObj, R, T0>::invoke(boost::detail::function::function_buffer&, T0) [with FunctionObj = boost::_bi::bind_t<void, boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>, boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> > >; R = void; T0 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >]’
/usr/include/boost/function/function_template.hpp:940:38:   required from ‘void boost::function1<R, T1>::assign_to(Functor) [with Functor = boost::_bi::bind_t<void, boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>, boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> > >; R = void; T0 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >]’
/usr/include/boost/function/function_template.hpp:728:7:   required from ‘boost::function1<R, T1>::function1(Functor, typename boost::enable_if_c<boost::type_traits::ice_not<boost::is_integral<Functor>::value>::value, int>::type) [with Functor = boost::_bi::bind_t<void, boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>, boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> > >; R = void; T0 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >; typename boost::enable_if_c<boost::type_traits::ice_not<boost::is_integral<Functor>::value>::value, int>::type = int]’
/usr/include/boost/function/function_template.hpp:1077:16:   required from ‘boost::function<R(T0)>::function(Functor, typename boost::enable_if_c<boost::type_traits::ice_not<boost::is_integral<Functor>::value>::value, int>::type) [with Functor = boost::_bi::bind_t<void, boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>, boost::_bi::list2<boost::_bi::value<joint_trajectory_execution::TrajectoryActionServer*>, boost::arg<1> > >; R = void; T0 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >; typename boost::enable_if_c<boost::type_traits::ice_not<boost::is_integral<Functor>::value>::value, int>::type = int]’
/home/phil/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/src/TrajectoryActionServer.cpp:270:26:   required from here
/usr/include/boost/bind/bind.hpp:313:35: error: no match for call to ‘(boost::_mfi::mf1<void, joint_trajectory_execution::TrajectoryActionServer, actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&>) (joint_trajectory_execution::TrajectoryActionServer*&, const actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&)’
         unwrapper<F>::unwrap(f, 0)(a[base_type::a1_], a[base_type::a2_]);
                                   ^
In file included from /usr/include/boost/bind/mem_fn.hpp:215:0,
                 from /usr/include/boost/mem_fn.hpp:22,
                 from /usr/include/boost/function/detail/prologue.hpp:18,
                 from /usr/include/boost/function.hpp:24,
                 from /opt/ros/kinetic/include/ros/forwards.h:40,
                 from /opt/ros/kinetic/include/ros/common.h:37,
                 from /opt/ros/kinetic/include/ros/ros.h:43,
                 from /home/phil/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/include/joint_trajectory_execution/TrajectoryActionServer.h:27,
                 from /home/phil/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/src/TrajectoryActionServer.cpp:19:
/usr/include/boost/bind/mem_fn_template.hpp:163:7: note: candidate: R boost::_mfi::mf1<R, T, A1>::operator()(T*, A1) const [with R = void; T = joint_trajectory_execution::TrajectoryActionServer; A1 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&] <near match>
     R operator()(T * p, A1 a1) const
       ^
/usr/include/boost/bind/mem_fn_template.hpp:163:7: note:   conversion of argument 2 would be ill-formed:
In file included from /usr/include/boost/bind.hpp:22:0,
                 from /opt/ros/kinetic/include/ros/publisher.h:35,
                 from /opt/ros/kinetic/include/ros/node_handle.h:32,
                 from /opt/ros/kinetic/include/ros/ros.h:45,
                 from /home/phil/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/include/joint_trajectory_execution/TrajectoryActionServer.h:27,
                 from /home/phil/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/src/TrajectoryActionServer.cpp:19:
/usr/include/boost/bind/bind.hpp:313:35: error: binding ‘const actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >’ to reference of type ‘actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&’ discards qualifiers
         unwrapper<F>::unwrap(f, 0)(a[base_type::a1_], a[base_type::a2_]);
                                   ^
In file included from /usr/include/boost/bind/mem_fn.hpp:215:0,
                 from /usr/include/boost/mem_fn.hpp:22,
                 from /usr/include/boost/function/detail/prologue.hpp:18,
                 from /usr/include/boost/function.hpp:24,
                 from /opt/ros/kinetic/include/ros/forwards.h:40,
                 from /opt/ros/kinetic/include/ros/common.h:37,
                 from /opt/ros/kinetic/include/ros/ros.h:43,
                 from /home/phil/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/include/joint_trajectory_execution/TrajectoryActionServer.h:27,
                 from /home/phil/catkin_ws/src/joint-control-pkgs/joint_trajectory_execution/src/TrajectoryActionServer.cpp:19:
/usr/include/boost/bind/mem_fn_template.hpp:168:25: note: candidate: template<class U> R boost::_mfi::mf1<R, T, A1>::operator()(U&, A1) const [with U = U; R = void; T = joint_trajectory_execution::TrajectoryActionServer; A1 = actionlib::ServerGoalHandle<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >&]
[... cut ...]
make[2]: *** [joint-control-pkgs/joint_trajectory_execution/CMakeFiles/trajectory_action_server.dir/src/TrajectoryActionServer.cpp.o] Error 1
CMakeFiles/Makefile2:1777: recipe for target 'joint-control-pkgs/joint_trajectory_execution/CMakeFiles/trajectory_action_server.dir/all' failed
make[1]: *** [joint-control-pkgs/joint_trajectory_execution/CMakeFiles/trajectory_action_server.dir/all] Error 2

I put the c++11 flags in the cmake files. It's weird because I managed to build it on previous builds but not this one. I'm sure I am missing a simple thing but i can't figure it out.

Thank you for your help!

wonwon0, 2018-01-25:
I just reformatted on a clean install, no bugs anymore.

closed 2018-01-25 by wonwon0

JenniferBuehler, 2018-01-26:
Sorry I saw this too late. Glad it has been resolved!