JenniferBuehler/gazebo-pkgs #15: Grasping an object without relying on ROS?

sahiljuneja, 2018-02-22:
Hey Jennifer, continuing our discussion from the email thread (thanks for the response on that!).

So, my question is how to utilize this plugin (or just the code) to allow for a manipulator to grasp an object, using a two-finger gripper only in gazebo.

I am trying to understand your other tutorials as well, but you seem to rely on moveit for the actual grasping, correct?

In gazebo, I can essentially set joint position values for two prismatic fingers of the gripper (assuming a simple scenario here) to allow the gripper to hold an object. Would I need your plugin in that case? I plan to have a case where the object pose is detected to be between the two gripper fingers, and if yes, then set the prismatic joint positions to close on the object. Which isn't as simple I think, it's more of defining the proper mathematical conditions to have the gripper in relation to the object pose in the right way. Can also assume of a simple case where two sides of the object ( in this case a cube) are parallel to the fingers.

So, I was hoping to get an idea from you how much of this can your plugin help achieve (and how without the ROS part)? [Hopefully, the above was clear]

JenniferBuehler, 2018-02-23:
Hi,
the Gazebo plugin actually only helps to keep the object in place in the gripper. Especially the older versions of Gazebo don't do splendidly for grasping objects. So you may find that once you have moved your grippers into place so they grasp the object, the object still slips off the robot's gripper, or it may even happen that the whole robot explodes and flies off into space :o because the forces applied to the object create an instabilities in the physics engine. So that's when the gazebo grasp plugin here comes into play.

The grasp plugin itself can be used with Gazebo only. Only the cmake build is currently laid out for catkin but you shouldn't need to change much to compile it stand-alone. I can help you with that, but let's first establish whether this is what you actually need.

No, these packages don't rely on MoveIt. MoveIt can help you to plan the movement required to get the object into place. The MoveIt-related code in my repositories that you refer to is probably the one in the grasp-execution-pkgs repository? That's all pretty tied to ROS at the moment I'm afraid, it would take a bit more effort to make this Gazebo-only (though it is possible).

I guess you checked out the [wiki of this repo](https://github.com/JenniferBuehler/gazebo-pkgs/wiki) already? It describes a bit more about what's in this repository.

sahiljuneja, 2018-02-23:
Aah, interesting to know. I came across the gripper.h files in the api so was wondering how yours is different. This is helpful to know!

I was able to compile it, I believe. Didn't face an issue with that I think (shared object file was created and added the plugin to my gazebo sdf/world file; doesn't throw any errors for that as of now). 

Yes, I was checking out the execution-pkgs repo as well. And I did check out the other wiki as well, specifically the https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin part which seemed relevant to me. My question was around that and the next step. 

As I said above, I managed to compile the plugin and added to my arm configuration in gazebo. What I am trying to narrow down is to have the gripper grasp an object. What exactly needs to be "triggered" for it to grasp a particular object, and how I can use your plugin to hold it in place and check if the object is being properly held by the gripper or not.

JenniferBuehler, 2018-02-23:
The trigger for the grasp is that opposing forces from the gripper act upon the object, then it gets attached automatically. You just have to set the parameters for your robot correctly, for this you have already found the right wiki page which explains the parameters.

Note that a few people recently had problems with their robots and the plugin, see issue #9. I'll be trying to fix this as soon as I get the time, hopefully very soon.

JenniferBuehler, 2018-02-23:
Ah, and to (3): There is currently no way to check this, though it would be easy to add a function to check for this condition to the plugin. However then there is the problem of how to access the plugin from your software. It would make sense to publish a message stream (can be a gazebo message) which sends a message each time the attachment status changes. But that would have to be implemented still.

The thing I don't like about that is that it's not something that mirrors the real robot - the real robot wouldn't be publishing such info, unless it had sensors in the grippers and then it would still need to run a node which checks for attachment. So to do it properly I'd rather implement something like that, e.g. another Gazebo plugin which simulates the robot's "gripper attachment checker" or something like that.

JenniferBuehler, 2018-02-26:
Yes, you've got it :)
The plugin is the same whether you have ROS or not. But ROS will make it easier with all the rest you need for your tasks, e.g. planning the arm movement with MoveIt! That said, you can do all of this using Gazebo only too, maybe you just need to do a bit more work on integrating 3rd party packages into your code first.

sahiljuneja, 2018-02-27:
Yeah, I might consider using a vacuum gripper instead maybe. That might be perhaps easier than a 2-finger approach, using a contact sensor plugin and creating a link between object and gripper. I guess will experiment with that. Thanks for all your help!

closed 2019-03-01 by JenniferBuehler