JenniferBuehler/gazebo-pkgs #16: The robot doesn't move according to the GUI(joint_state_publisher)

Sowjanya06, 2018-03-12:
Hi Jennifer,
firstly, thank you very much for this much need package.
I am very much new to ROS and Gazebo. I have tried to follow this  [https://github.com/JenniferBuehler/joint-control-pkgs/wiki/gazebo_joint_control](url) and service request
 to gazebo to set the physics properties was required , I have cloned the gazebo-pkgs into my catkin ws. But, i see that my simple robot is not able to move with the GUI. or may be i cannot understand how to integrate all of this. I am attaching the screenshot of my robot which has only one movable joint. It can be an overkilling of such huge package to this bot, But I just want to learn how it works
It would be of great help if you can look into this matter.
THANKS FOR READING
[image]

JenniferBuehler, 2018-03-13:
Hi, glad this package is helping :) It's a bit hard to tell why your robot is not moving without knowing the exact set-up you have. Did you set up the [arm components manager](https://github.com/JenniferBuehler/convenience-pkgs/wiki/Arm-components-name-manager)? Can you send the output of the terminal?

The easiest would be if you can share your code so I can look at which configuration is missing or what else is going wrong.

Sowjanya06, 2018-03-14:
Hi, I am glad that you responded !this is what my terminal says

... logging to /home/lakshmic/.ros/log/ce4501f0-273e-11e8-9656-f04da2da84a0/roslaunch-lakshmi-c-3892.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lakshmi-c:39299/

SUMMARY
========

PARAMETERS
 * /arm/arm_effort_controller_names: ['link2_link3_eff...
 * /arm/arm_joint_init: [0, 0, 0, 1.572]
 * /arm/arm_joint_max_force: [50, 50, 50, 50]
 * /arm/arm_joints: ['base_footprint_...
 * /arm/arm_links: ['base_footprint'...
 * /arm/joint_state_controller/publish_rate: 100
 * /arm/joint_state_controller/type: joint_state_contr...
 * /arm/link2_link3_effort_controller/joint: link2_link3_joint
 * /arm/link2_link3_effort_controller/pid/d: 0
 * /arm/link2_link3_effort_controller/pid/i: 0.1
 * /arm/link2_link3_effort_controller/pid/p: 20
 * /arm/link2_link3_effort_controller/type: effort_controller...
 * /arm/palm_link: link3
 * /joint_state_control_topic: arm/joint_control
 * /joint_state_publisher/rate: 10
 * /joint_state_publisher/use_gui: True
 * /preserve_original_angles: link2_link3_joint
 * /publish_joint_states_topic: arm/joint_state
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 100.0
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    urdf_spawner (gazebo_ros/spawn_model)
  /arm/
    controller_spawner (controller_manager/spawner)

auto-starting new master
process[master]: started with pid [3907]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ce4501f0-273e-11e8-9656-f04da2da84a0
process[rosout-1]: started with pid [3920]
started core service [/rosout]
process[gazebo-2]: started with pid [3923]
process[gazebo_gui-3]: started with pid [3927]
process[urdf_spawner-4]: started with pid [3931]
process[robot_state_publisher-5]: started with pid [3934]
process[joint_state_publisher-6]: started with pid [3938]
process[arm/controller_spawner-7]: started with pid [3940]
[ INFO] [1521001145.427973149]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1521001145.439326793]: Finished loading Gazebo ROS API Plugin.
[INFO] [WallTime: 1521001147.466754] [0.000000] Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1521001148.453519931, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1521001148.595489911, 0.162000000]: Physics dynamic reconfigure ready.
[urdf_spawner-4] process has finished cleanly
log file: /home/lakshmic/.ros/log/ce4501f0-273e-11e8-9656-f04da2da84a0/urdf_spawner-4*.log

JenniferBuehler, 2018-03-14:
Hi,

so it looks like you did set up the arm component manager, given that it matches the joint names in your URDF, that's probably not the issue.

I will look at this in the next few days when I got a bit of time in-between other commitments. If you can share your repository with me to look at your setup, that would be easier, but if not then I'll build another small example which you can then follow. I looked at the tutorial and it probably can use a make-over anyway, it's been a while since it's been polished ;)

Sowjanya06, 2018-03-14:
Hi, 
yeah i can surely share, how should I share my package with you? May I have your mail id.

JenniferBuehler, 2018-03-15:
I've updated [the wiki](https://github.com/JenniferBuehler/joint-control-pkgs/wiki/gazebo_joint_control) and made an [example for a simplified version of the RRBOT](https://github.com/JenniferBuehler/joint-control-pkgs/tree/master/gazebo_joint_control/rrbot_example) from the gazebo tutorials.

``roslaunch gazebo_joint_control rrbot_simplified_gazebo.launch``

It's only a very quick update, not that polished yet, but maybe this simple example helps you to find the difference in your setup. It's much simpler than the Jaco example mentioned in the wiki before.

You will need to do a new pull of convenience-pkgs and of this repo for everything to work (update in convenience-pkgs now allows a setup of an arm without a gripper).

Let me know if this helped you to find the issue.

Sowjanya06, 2018-03-15:
Thank You very much Jennifer
I will let you know the result very shortly.

JenniferBuehler, 2019-03-01:
I'm guessing since this is quite long ago, it is no longer relevant.
If anyone experiences a similar issue, please re-open this.

closed 2019-03-01 by JenniferBuehler