JenniferBuehler/gazebo-pkgs #20: Grasping doesn't occurs while robot takes detail.

Stayermax, 2018-05-24:
Hi, Jennifer!

I've got problem with grasping an object, since I think grasp-fix plugin was installed incorrectly. I already tried to reinstall it a few times' but robot still doesn't grasp an object. Can you please take a look?

I don't get any messages about grasping fix and when robot takes the cube, there is not INFO message about grasping also.

[image]
[image]
[image]

JenniferBuehler, 2018-05-25:
Hi, 
sorry you're having trouble with this. It doesn't seem to load the plugin. You should at least see the message "Loading grasp-fix plugin". I take it you re-compiled your catkin workspace a few times? Are you loading the ``devel/setup.bash`` either manually or from your ``.bashrc``? All you other programs in your catkin workspace can be run, e.g. by ``rosrun`` or ``roslauch``?

Stayermax, 2018-05-25:
Yes, I'm loading setup.bash, so If in terminal I enter roslaunch and press TAB, terminal gives me an opportunity to choose gazebo_grasp_plugin, but I can not run it with roslaunch or with rosrun:
[image]

Maybe I should add a plugin using gazebo_world_plugin_loader?

JenniferBuehler, 2018-05-27:
Hi,
the gazebo_grasp_plugin is a Gazebo model plugin, so you cannot start it. And because it is a model plugin, it should automatically be loaded when you load your robot URDF, you cannot use it in the gazebo_world_plugin_loader.

This should be done automatically, but maybe you need to add the library to the ``GAZEBO_PLUGIN_PATH`` explicitly.

Run this in your terminal where you start up the gazebo node:

``export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:`locate libgazebo_grasp_fix.so` ``

Does it work with that?

If this still doesn't work, another quick way to check if the problem is the library not being found:
introduce a typo into your URDF file (e.g. "libgazebo_NO_grasp_fix.so"), and see if any error is printed. You can also introduce other stupid errors into the URDF/XACRO file and see if the errors are printed as expected. If obious errors are not printed, then for some reason it is not loading  your URDF with the grasp fix.

Stayermax, 2018-05-31:
Hi, 
if I run this command:
`export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH: 'locate libgazebo_grasp_fix.so'`
terminal tells me something like this:
[image]

Also changing of URDF or XACRO files doesn't give effect. I think it's because I connect to robot files that are in the cloud of the laboratory and I can't change them, since I'm not the administrator of the lab. 

But I solved my problem of cube dropping, by changing way of robot work, so now robot's arm doesn't shake and hence detail is firmly fixed in the gripper.

So thank you for your help!

JenniferBuehler, 2018-06-02:
Ok, glad you resolved the problem in other ways.
That it references the library file in some "trash" folder seems a bit suspicious, maybe it's got to do with that. But in any case, now that you've solved it it doesn't matter any more ;)

Cheers!

closed 2018-06-02 by JenniferBuehler