JenniferBuehler/gazebo-pkgs #21: Is there a version that works with Gazebo 9?

Shreeyak, 2018-07-01:
I was working with ROS Kinetic and had Gazebo 9 installed. Upon trying to compile gazebo-grasp-plugin, it gave me various errors. Seemed like a bunch of functions from gazebo 7 were not available in gazebo 9. Is there a new version of these packages that supports gazebo 9? 

THe newest ROS melodic also uses gazebo 9. Anyone tried this with ros melodic?

JenniferBuehler, 2018-07-03:
I am intending to upgrade this to be compatible with melodic and also create a ros-free version, however in the next few weeks it's not looking like I will get much time to do this.

If you want to submit a PR with a ros-free version for Gazebo 9 I'll be happy to review it and merge it.

Or I'll let you know here once I get around to do it myself, though it could take a few weeks.

JenniferBuehler, 2019-02-27:
There is a pull request #23 for this now, this will be merged very soon, so closing this issue.

closed 2019-02-27 by JenniferBuehler

Shreeyak, 2019-03-04:
Oh wow, that's great! Thanks!