JenniferBuehler/gazebo-pkgs #22: Getting the grasp-plugin to grip multiple bodies

zighi, 2018-08-07:
Hello,
I'm trying to use the plugin in model to move a stack of discs, where the robot is supposed to hold and pick only the one on the bottom. The discs on the top however don't move horizontally with the gripped disc and slip off(I will comment a video as soon as I can). I tried to solve the problem through friction parameters but had no success. I was wondering if there is a way I could modify the plugin, so that all the discs are bound to the gripper link. Since I'm a beginner in Gazebo I don't know what other options are there. Any tipps or ideas are appreciated.

JenniferBuehler, 2018-08-07:
Hi,

it should in theory be possible to grasp multiple bodies, though admittedly I haven't tested it specifically, I just wrote the code so that multiple objects can be handled (see [GazeboGraspFix](https://github.com/JenniferBuehler/gazebo-pkgs/blob/master/gazebo_grasp_plugin/include/gazebo_grasp_plugin/GazeboGraspFix.h) where all relevant fields are indexed by the object name). 

When you tried it, what happened? Did it only grasp the top disc? A video would be helpful :)

Cheers
Jennifer

zighi, 2018-08-08:
Hi,

I've uploaded a video and a log File in case the GIF wasn't clear. I'm still studying up on plugins and still haven't figured out how the methods in the Gazebograsp plugin are called during the simulation and where exactly it's decided if and object is gripped, so i couldn't change anything for a test, so I asked the question here.

Thanks for the help.

best regards

[image]
[sliding-discs.log](https://github.com/JenniferBuehler/gazebo-pkgs/files/2270005/sliding-discs.log)

JenniferBuehler, 2018-08-09:
Ok, so from what I can see in the video, the problem you're having is an entirely different one.
The grasp plugin holds for the lowest disc. It will not have effect on the two top discs because the gripper is not even touching them. However, all discs should still be lifted if you are anyway grasping the bottom one.

What happens here is that the top discs "slide off" as the robot moves sideways. It looks to me like the friction of the discs is 0, so the friction of the bottom disc is not enough to "drag along" the smaller discs with it. Of course, that's only a guess because I can't see your model...  one thing that is weird is the way how they then fall on the table so neatly. Is it possible that you update the disc positions from somewhere else?

zighi, 2018-08-16:
I have not described the problem perfectly but I'm glad the video is clear. The urdf of the discs looks something like, what I've attached as txt. Apart from the size, all the discs are similar. 
I've tweaked the parameters "mu", "fdir" , "minDepth" and "maxVel" a lot, but couldn't get an acceptable behavior from the discs. With the parameters set below, when I exert a force manually (in gazebo GUI, right click on model -> Apply Force), the top discs drag along a little (still slide off after a while even though the friction coefficient is set unrealistically high), but they slide off instantly when moved by the robot., that's why I thought of the dirty workaround of fixing the top discs to the bottom one.
about positioning the discs, I can only spawn the discs on different positions but afterwards, I can only move them by applying a force on them.

I really appreciate the help
Best regards

[disc1.urdf.txt](https://github.com/JenniferBuehler/gazebo-pkgs/files/2293015/disc1.urdf.txt)

JenniferBuehler, 2018-08-23:
Ok, so you tried the friction parameters. Gazebo isn't perfect with simulation like in this case... however I don't see how this could be fixed directly in the grasp plugin, because the robot is not really touching the discs. What you could do is to write a separate plugin which checks which objects are stacked on each other, and then attach them to each other. Then when the bottom disc is grabbed, the others will move along too.
Another solution could be to publish a gazebo message from the grasp plugin each time a grasp attachment has happened. You could subscribe to this topic and each time an attachment has happened, fix all stacked objects on the grasped object. And then detach them, when the "detach" message arrived. This would probably be the nicer way to do it ;)

I don't have time at the moment to do this, but PR's are welcome with the message publishing feature. Or if  you have time to wait a few weeks, I could then have the time to do it myself.

zighi, 2018-08-24:
I'll start working on the idea, but I can't estimate how long it will take me to get it done.
At the same time I'll try to find a solution through correcting the friction between the discs.
I'll post the solution here when I find it.
Best regards

zighi, 2018-09-28:
Just an update, the problem was solved without the need to write a plugin. The problem was solved by controlling the robot through EffortJointInterface instead of PositionJointInterface. From what I understood, when using PositionJointInterface forces wont be relevant on the positioning of the end effector (and the gripped object), therefore the friction forces are ignored on the dynamic of the objects.
Solution to a similar problem : http://answers.gazebosim.org/question/16928/friction-not-working-between-urdf-and-database-model/

Thanks for the tips
Best regards

JenniferBuehler, 2018-09-29:
Ah, awesome! I didn't think about the controls of the robot in the first place...
Thanks for the update and the link.
I guess that means I can close this for now :)

closed 2018-09-29 by JenniferBuehler