JenniferBuehler/gazebo-pkgs #24: catkin_make error because of protoc

MaSchle, 2018-08-29:
Hello,
after cloning the package into my catkin_ws I receive multiple errors during the execution of catkin_make, which all seem to be related to gazebo.
Type 1:
[image]

Although my version of **protobuf** should be the right one:
[image]
I already tried to regenerate the files as stated in the error message with the following command in the respective directory:
`protoc *.proto --cpp_out=..`
But it didn't resolve the problem.

Type 2:
[image]

Type 3:
[image]

The 3 screenshots are only excerpts of the whole command line output. In reality there are much more error messages, but all of the same type as the abovementioned.

### My system:
Ubuntu 16.04
ROS Kinetic
Gazebo 7.14.0
I installed gazebo and the respective gazebo_ros packages via apt-get.

JenniferBuehler, 2018-08-31:
Hmmm ok, thanks for reporting this. I'll check as soon as I can if I can reproduce this. I'm currently on an old system with indigo so can't check quickly, after I reboot into Xenial next time I'll check on this.

From how I understand these messages, it sounds like your protoc version is too new for the Gazebo 7 files. Have you tried to make a simple dummy catkin package with just a simple .cpp file which includes `gazebo.hh`? This would be worth a try to see if you get the same error then.

JenniferBuehler, 2019-03-01:
I've had the same issue on Bionic/Melodic. It goes away after re-compling a 2nd time. It happens every time some CMakeLists.txt has changed so my guess is that some dependencies in message building haven't been met. I haven't yet had time to look into this, until then the best fix of this is to just re-compile.

JenniferBuehler, 2019-03-01:
Actually, sorry I misread the error messages above, they are slightly different. I've had other ones. I've recently compiled in Kinetic and also Melodic and haven't encountered the same issue.
If this is still relevant at all, you could try to pull the new version of my packages, they have just been updated.

JenniferBuehler, 2019-05-03:
I haven't had the issue again, please re-open if this still happens to you.

closed 2019-05-03 by JenniferBuehler