JenniferBuehler/gazebo-pkgs #25: Grasp plugin doesn't work with a gazebo issue

saikishor, 2018-09-19:
There is a gazebo issue that makes the gripper open and close due to an unknown reason. I thought this plugin would fix that, but as the arm opens and closes, the object is falling down. Do you know any workaround?.

Thank you,

JenniferBuehler, 2018-09-25:
This plugin is meant to release the object once the gripper is not touching it any more. So it's not suitable to fix this bug. Is there an issue in Gazebo for this bug?

saikishor, 2018-09-25:
@JenniferBuehler  No I don't think there is such issue on this bug. I don't know why it is happening, but it is totally strange as the same thing worked in the Gazebo 2.2.3, but now when upgraded to kinetic we use gazebo 7.12.0 here comes the problem of the gripper, when doing a motion after grasping the object. I created a question in the gazebo answers(http://answers.gazebosim.org/question/20490/issue-with-the-post-grasp-motion-in-gazebo/). I recently created an issue on gazebo(https://bitbucket.org/osrf/gazebo/issues/2528/issue-with-the-post-grasp-motion-with-the)

JenniferBuehler, 2018-09-26:
Ok, thanks for the links. Let's see if there is a response to the issue soon.
Meanwhile, I'll close this, because the grasp plugin unfortunately cannot be used to fix the issue.
Cheers

closed 2018-09-26 by JenniferBuehler