JenniferBuehler/gazebo-pkgs #28: gzserver: symbol lookup error

rdesc, 2019-03-05:
Hi,

I have experienced the following error which occurs for all commits after 22d171b2eaa3ad6a340b892394fe949341669d46. I am running ROS kinetic on a Ubuntu 16.04.6 docker container.

`````````````````````````````````
[Msg] Loading grasp-fix plugin
[Msg] GazeboGraspFix: Using disable_collisions_on_attach 0
[Msg] GazeboGraspFix: Using update rate 4
[Msg] GazeboGraspFix: Using max_grip_count 8
[Msg] GazeboGraspFix: Using grip_count_threshold 4
[Msg] GazeboGraspFix: Using release_tolerance 0.005
gzserver: symbol lookup error: /home/ros/catkin_ws/devel/lib/libgazebo_grasp_fix.so: undefined symbol: _ZN6gazebo10GetPhysicsERKN5boost10shared_ptrINS_7physics5WorldEEE
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/controller_manager", line 58, in <module>
    controller_manager_interface.load_controller(c)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/controller_manager/controller_manager_interface.py", line 58, in load_controller
    resp = s.call(LoadControllerRequest(name))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
Traceback (most recent call last):
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 207, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 185, in main
    resp = load_controller(name)
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 207, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 185, in main
    resp = load_controller(name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
      File "/opt/ros/kinetic/lib/controller_manager/spawner", line 207, in <module>
return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
        return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
if __name__ == '__main__': main()
    raise ServiceException("transport error completing service call: %s"%(str(e)))
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 199, in main
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
    resp = switch_controller(loaded, [], 2)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
    raise ServiceException("transport error completing service call: %s"%(str(e)))
`````````````````````````````````

JenniferBuehler, 2019-03-05:
Hi, this should have been fixed with a commit today, when was the last time you pulled?

rdesc, 2019-03-05:
Yes I tried with the latest commit (e54939f6a80982dc1b89c3c2fb288e989f758b20).

JenniferBuehler, 2019-03-05:
Hmm weird, in issue #26 the newest update resolved the problem.
Can you please post the output of
``ldd ` locate libgazebo_grasp_fix.so` | grep gazebo``
Thank you :)

rdesc, 2019-03-20:
@JenniferBuehler never mind all good it now works thank you!

closed 2019-03-20 by rdesc