JenniferBuehler/gazebo-pkgs #31: GazeboGraspGripper: Palm link ee_link not found. The gazebo grasp plugin will not work.

gtatiya, 2019-05-04:
Hi JenniferBuehler,

Thanks for creating GazeboGraspFix plugin. I'm trying to use it with UR5 arm and RobotiQ 85 gripper. But, the object falls as before and I get the following error:

```
Loading grasp-fix plugin
GazeboGraspFix: Using disable_collisions_on_attach 1
GazeboGraspFix: Using update rate 2
GazeboGraspFix: Using max_grip_count 2
GazeboGraspFix: Using grip_count_threshold 1
GazeboGraspFix: Using release_tolerance 0.005
[Err] [GazeboGraspGripper.cpp:59] GazeboGraspGripper: Palm link ee_link not found. The gazebo grasp plugin will not work.
[Err] [GazeboGraspFix.cpp:173] GazeboGraspFix: Could not initialize arm gripper_arm. Skipping.
[Err] [GazeboGraspFix.cpp:196] ERROR: GazeboGraspFix: Cannot use a GazeboGraspFix because no arms were configured successfully. Plugin will not work.
Loading grasp-fix plugin
GazeboGraspFix: Using disable_collisions_on_attach 0
GazeboGraspFix: Using update rate 4
GazeboGraspFix: Using max_grip_count 8
GazeboGraspFix: Using grip_count_threshold 4
GazeboGraspFix: Using release_tolerance 0.005
GazeboGraspFix: Adding collision scoped name robot::gripper_finger1_finger_tip_link::gripper_finger1_finger_tip_link_collision
GazeboGraspFix: Adding collision scoped name robot::gripper_finger1_inner_knuckle_link::gripper_finger1_inner_knuckle_link_collision
GazeboGraspFix: Adding collision scoped name robot::gripper_finger1_knuckle_link::gripper_finger1_knuckle_link_collision
GazeboGraspFix: Adding collision scoped name robot::gripper_finger1_knuckle_link::gripper_finger1_knuckle_link_fixed_joint_lump__gripper_finger1_finger_link_collision_1
GazeboGraspFix: Adding collision scoped name robot::gripper_finger2_finger_tip_link::gripper_finger2_finger_tip_link_collision
GazeboGraspFix: Adding collision scoped name robot::gripper_finger2_inner_knuckle_link::gripper_finger2_inner_knuckle_link_collision
GazeboGraspFix: Adding collision scoped name robot::gripper_finger2_knuckle_link::gripper_finger2_knuckle_link_collision
GazeboGraspFix: Adding collision scoped name robot::gripper_finger2_knuckle_link::gripper_finger2_knuckle_link_fixed_joint_lump__gripper_finger2_finger_link_collision_1

```

This is how my gzplugin_grasp_fix.urdf.xacro file looks like:
```
<?xml version="1.0" encoding="UTF-8"?>
<root 
xmlns:sensor="http://playerstage.sourceforge.net/gazebo/xmlschema/#sensor"
xmlns:controller="http://playerstage.sourceforge.net/gazebo/xmlschema/#controller"
xmlns:interface="http://playerstage.sourceforge.net/gazebo/xmlschema/#interface"
xmlns:xacro="http://ros.org/wiki/xacro">

<xacro:macro name="gzplugin_grasp_fix">
    <gazebo>
        <plugin name="gazebo_grasp_fix" filename="libgazebo_grasp_fix.so">
            <arm>
                <arm_name>UR5Arm</arm_name>
                <palm_link>wrist_3_link</palm_link>
                <gripper_link>gripper_finger1_inner_knuckle_link</gripper_link>
                <gripper_link>gripper_finger1_finger_tip_link</gripper_link>
                <gripper_link>gripper_finger1_knuckle_link</gripper_link>
                <gripper_link>gripper_finger2_inner_knuckle_link</gripper_link>
                <gripper_link>gripper_finger2_finger_tip_link</gripper_link>
                <gripper_link>gripper_finger2_knuckle_link</gripper_link>
            </arm>
            <forces_angle_tolerance>100</forces_angle_tolerance>
            <update_rate>4</update_rate>
            <grip_count_threshold>4</grip_count_threshold>
            <max_grip_count>8</max_grip_count>
            <release_tolerance>0.005</release_tolerance>
            <disable_collisions_on_attach>false</disable_collisions_on_attach>
            <contact_topic>__default_topic__</contact_topic>
        </plugin>
    </gazebo>
</xacro:macro>
</root>
```

Here's a video: https://drive.google.com/file/d/1pvHDf6qhK75cAdyv3fhh6LAzdl0HAG85/view?usp=sharing

You can run it yourself by cloning this repo: https://github.com/gtatiya/Hacking-SotA-UR5 and running:
```
roslaunch ur5_single_arm_manipulation pick_and_place.launch 
roslaunch ur5_single_arm_manipulation grasp_generator_server.launch 
rosrun ur5_single_arm_manipulation pick_and_place.py
```

**How can I fix this error?**

closed 2019-05-05 by gtatiya

JenniferBuehler, 2019-05-06:
I guess you fixed this, since it's been closed?

truhoang-ubtrobot, 2020-01-24:
Just want to post an answer here if people runs into this issue: Gazebo combines all links connected by fixed joints together to reduce computational complexity so the link specified in `<palm_link>` or `<gripper_link>` may not exist in the model Gazebo has of your robot (answered [here](https://answers.ros.org/question/263398/continuous-joint-will-not-load-in-gazebo/)). In the OP's case, replacing `wrist_3_link` with `gripper_finger1_inner_knuckle_link` works:

```
<palm_link> gripper_finger1_inner_knuckle_link </palm_link>
<gripper_link> gripper_finger1_finger_tip_link </gripper_link>
<palm_link> gripper_finger2_inner_knuckle_link </palm_link>
<gripper_link> gripper_finger2_finger_tip_link </gripper_link>
```

anubhav1772, 2020-09-15:
Does it also work with **kinetic** and **gazebo 9**, since in my case it's not working?

youngblood1998, 2020-12-09:
hi，I recently meet the same  problem，if you fixed your error，can you tell me how to do，thank you very much，it is important for me。

JenniferBuehler, 2020-12-09:
Did you try the solution posted by @truhoang-ubtrobot?
Are you seeing any console output from the plugin? If unsure, switch on gazebo verbose output (in the launch file which brings up gazebo node), it could be the plugin isn't loaded when the library wasn't found, unfortunately you'll only see this with verbose mode.

JenniferBuehler, 2020-12-09:
See also [this wiki](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/Troubleshooting)

youngblood1998, 2020-12-10:
@JenniferBuehler Thank you for your reply,I had solved my problem yesterday,I add the plugin in the xacro file of my robot and finally it can be loaded successfully.

GradyM2M, 2021-04-23:
@JenniferBuehler   Hi JenniferBuehler, I encountered some problems in using the gazebograspfix plugin. Could you explain the specific meaning of the parameters in the following tags?
<forces_angle_tolerance>
<grip_count_threshold>
<max_grip_count>
<release_tolerance>
<disable_collisions_on_attach>
Thanks!

JenniferBuehler, 2021-04-23:
Hello, they are all explained in [the wiki](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin)

getupcurry, 2021-07-25:
> @JenniferBuehler Thank you for your reply,I had solved my problem yesterday,I add the plugin in the xacro file of my robot and finally it can be loaded successfully.

Hi, how did you solve the problem?

ProfJust, 2022-03-09:
Finally fixed grasping works for the UR5 with robotiq_85_gripper.

Change file  <common.gazebo.xacro>  in folder universal_robot/ur_description/urdf to this

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://wiki.ros.org/xacro">

  <gazebo>
    <plugin name="ros_control" filename="libgazebo_ros_control.so">
    </plugin>
    
    <!-- https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin -->
    <plugin name="gazebo_grasp_fix" filename="libgazebo_grasp_fix.so">
        <arm>
           <arm_name>ur5</arm_name>  <!-- name of this arm. Has to be unique.-->

           <!--  has to be the link to which the finger joints are attached. -->
            <palm_link>robotiq_85_left_inner_knuckle_link</palm_link>
            <palm_link>robotiq_85_right_inner_knuckle_link</palm_link>
           <!-- tags have to include all link names of the gripper/hand which are used to actively grasp objects -->           
              <gripper_link>robotiq_85_left_finger_tip_link</gripper_link>
              <gripper_link>robotiq_85_right_finger_tip_link</gripper_link>          
       </arm>
       <forces_angle_tolerance>90</forces_angle_tolerance>
       <update_rate>4</update_rate>
       <grip_count_threshold>4</grip_count_threshold>
       <max_grip_count>8</max_grip_count>
       <release_tolerance>0.003</release_tolerance>
       <disable_collisions_on_attach>false</disable_collisions_on_attach>
       <contact_topic>__default_topic__</contact_topic>
    </plugin>
  </gazebo>
</robot>
```

JenniferBuehler, 2022-03-11:
Awesome, thank you for posting the solution!

tattrongvu, 2022-04-04:
Anyone else could run this in ROS Melodic, Ubuntu 18, and Gazebo 9x? I ran into the same problem and no method above could solve this. 
I use UR5e and robotiq 2f85.
This is my xacro file:

> <?xml version="1.0" encoding="UTF-8"?>

> <robot xmlns:xacro="http://www.ros.org/wiki/xacro" name = 'grasp fix'>
> 	
> 	<xacro:macro name="gzplugin_grasp_fix">
> 		<gazebo>
> 			<plugin name="gazebo_grasp_fix" filename="libgazebo_grasp_fix.so">
> 					<arm>
> 							<arm_name>myur5e_fix</arm_name>
> 							<palm_link> gripper_finger1_inner_knuckle_link </palm_link>
> 							<palm_link> gripper_finger2_inner_knuckle_link </palm_link>
> 							<gripper_link>gripper_finger1_finger_tip_link</gripper_link>
> 							<gripper_link>gripper_finger2_finger_tip_link</gripper_link>
> 					</arm>
> 					<forces_angle_tolerance>100</forces_angle_tolerance>
> 					<update_rate>10</update_rate>
> 					<grip_count_threshold>1</grip_count_threshold>
> 					<max_grip_count>3</max_grip_count>
> 					<release_tolerance>0.0198</release_tolerance>
> 					<disable_collisions_on_attach>false</disable_collisions_on_attach>
> 					<contact_topic>__default_topic__</contact_topic>
> 			</plugin>
> 		</gazebo>
> 	</xacro:macro>
> </robot>

The errors:

> [ INFO] [1649032104.508383661, 0.273000000]: Loaded gazebo_ros_control.
> [Msg] Loading grasp-fix plugin
> [Msg] GazeboGraspFix: Using disable_collisions_on_attach 1
> [Msg] GazeboGraspFix: Using update rate 2
> [Msg] GazeboGraspFix: Using max_grip_count 2
> [Msg] GazeboGraspFix: Using grip_count_threshold 1
> [Msg] GazeboGraspFix: Using release_tolerance 0.005
> [Err] [GazeboGraspGripper.cpp:64] GazeboGraspGripper: Palm link wrist_attach not found. The gazebo grasp plugin will not work.
> [Err] [GazeboGraspFix.cpp:231] GazeboGraspFix: Could not initialize arm gripper_arm. Skipping.
> [Err] [GazeboGraspFix.cpp:261] ERROR: GazeboGraspFix: Cannot use a GazeboGraspFix because no arms were configured successfully. Plugin will not work.
> [ INFO] [1649032104.520784734, 0.273000000]: MimicJointPlugin loaded! Joint: "gripper_finger1_joint", Mimic joint: "gripper_finger2_joint", Multiplier: 1, Offset: 0, MaxEffort: 1000, Sensitiveness: 0
> [ INFO] [1649032104.521005784, 0.273000000]: MimicJointPlugin loaded! Joint: "gripper_finger1_joint", Mimic joint: "gripper_finger2_inner_knuckle_joint", Multiplier: 1, Offset: 0, MaxEffort: -1, Sensitiveness: 0
> [ INFO] [1649032104.521148414, 0.273000000]: MimicJointPlugin loaded! Joint: "gripper_finger1_joint", Mimic joint: "gripper_finger1_inner_knuckle_joint", Multiplier: 1, Offset: 0, MaxEffort: -1, Sensitiveness: 0
> [ INFO] [1649032104.521277124, 0.273000000]: MimicJointPlugin loaded! Joint: "gripper_finger1_joint", Mimic joint: "gripper_finger1_finger_tip_joint", Multiplier: -1, Offset: 0, MaxEffort: -1, Sensitiveness: 0
> [ INFO] [1649032104.521379014, 0.273000000]: MimicJointPlugin loaded! Joint: "gripper_finger1_joint", Mimic joint: "gripper_finger2_finger_tip_joint", Multiplier: -1, Offset: 0, MaxEffort: -1, Sensitiveness: 0
> [Msg] Loading grasp-fix plugin
> [Msg] GazeboGraspFix: Using disable_collisions_on_attach 0
> [Msg] GazeboGraspFix: Using update rate 10
> [Msg] GazeboGraspFix: Using max_grip_count 3
> [Msg] GazeboGraspFix: Using grip_count_threshold 1
> [Msg] GazeboGraspFix: Using release_tolerance 0.0198
> [Msg] GazeboGraspFix: Adding collision scoped name robot::gripper_finger1_finger_tip_link::gripper_finger1_finger_tip_link_collision
> [Msg] GazeboGraspFix: Adding collision scoped name robot::gripper_finger1_inner_knuckle_link::gripper_finger1_inner_knuckle_link_collision
> [Msg] GazeboGraspFix: Adding collision scoped name robot::gripper_finger2_finger_tip_link::gripper_finger2_finger_tip_link_collision
> [Msg] GazeboGraspFix: Adding collision scoped name robot::gripper_finger2_inner_knuckle_link::gripper_finger2_inner_knuckle_link_collision
> [Msg] Subscribing contact manager to topic ~/robot/contacts
> [Msg] Advertising grasping events on topic grasp_events
> Loaded 'joint_state_controller'
> Loaded 'myur5e_arm_controller'
> Loaded 'my2f85_gripper_controller'
> Started ['joint_state_controller'] successfully
> Started ['myur5e_arm_controller'] successfully
> Started ['my2f85_gripper_controller'] successfully
> [ INFO] [1649032105.082286141, 0.812000000]: Added FollowJointTrajectory controller for myur5e_arm_controller
> [robot_controllers-5] process has finished cleanly
> log file: /home/trong/.ros/log/21f7c634-b3ae-11ec-a73c-024244c36776/robot_controllers-5*.log
> [ INFO] [1649032105.388702212, 1.121000000]: Added FollowJointTrajectory controller for my2f85_gripper_controller
> [ INFO] [1649032105.388820832, 1.122000000]: Returned 2 controllers in list
> [ INFO] [1649032105.398784721, 1.133000000]: Trajectory execution is managing controllers
>

JenniferBuehler, 2022-04-05:
I cannot see `wrist_attach` in your list of links. you must give it the correct name in the URDF (gazebo plugin config part, see wiki)

tattrongvu, 2022-04-06:
> I cannot see `wrist_attach` in your list of links. you must give it the correct name in the URDF (gazebo plugin config part, see wiki)
Here is a path where i attach the robotiq gripper in to wrist attach in my urdf file
<xacro:robotiq_85_gripper prefix="" parent="wrist_attach" >
    <origin xyz="${wrist_length} 0 0" rpy="0 0 0"/>
  </xacro:robotiq_85_gripper>

Basically it is the same as @gtatiya since I take a robotiq description from him and add it to wrist_attach instead of ee_link. So based on the answer above and your description in the wiki I believe that <palm_link> should be the nearest unmoveable link with the _finger_tip_link (which will touch directly with the object). I have already try to open gazebo gui and see what link should be add and so i also tried to add "wrist_3_link" as palm link since it's the link in gazebo show that it connected togethers. I also tried add all of this like that

<palm_link> wrist_3_link</palm_link>
<palm_link> wrist_attach </palm_link>
<palm_link> gripper_finger1_inner_knuckle_link </palm_link>
<gripper_link>gripper_finger1_finger_tip_link</gripper_link>
<palm_link> gripper_finger2_inner_knuckle_link </palm_link>
<gripper_link>gripper_finger2_finger_tip_link</gripper_link>

Did i miss something?

Here is my urdf:

> <?xml version="1.0"?>
> <robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="myur5e" >
> 
>   
>   <xacro:include filename="$(find ur_e_description)/urdf/common.gazebo.xacro" />
> 
>   
>   <xacro:include filename="$(find ur_e_description)/urdf/ur5e.urdf.xacro" />
>   
>   
>   
>   <xacro:ur5e_robot prefix="" joint_limited="true"
>     shoulder_pan_lower_limit="${-pi}" shoulder_pan_upper_limit="${pi}"
>     shoulder_lift_lower_limit="${-pi}" shoulder_lift_upper_limit="${pi}"
>     elbow_joint_lower_limit="${-pi}" elbow_joint_upper_limit="${pi}"
>     wrist_1_lower_limit="${-pi}" wrist_1_upper_limit="${pi}"
>     wrist_2_lower_limit="${-pi}" wrist_2_upper_limit="${pi}"
>     wrist_3_lower_limit="${-pi}" wrist_3_upper_limit="${pi}"
>   />
> 
>   
>   <xacro:include filename="$(find robotiq_85_description)/urdf/robotiq_85_gripper.urdf.xacro" />
> 
>   
>   
>   <xacro:include filename="$(find myur5e_description)/urdf/mycamera.urdf.xacro" />
>   <xacro:mywristcamera parent="ee_link" >
>     <origin xyz="0 0 0" rpy="0 0 0"/>
>   </xacro:mywristcamera>
> 
>   <xacro:robotiq_85_gripper prefix="" parent="wrist_attach" >
>     <origin xyz="${wrist_length} 0 0" rpy="0 0 0"/>
>   </xacro:robotiq_85_gripper>
>   
> 
>   
>   <xacro:include filename="$(find myur5e_description)/urdf/mypedestal.urdf.xacro" />
> 
>   
>   <xacro:include filename="$(find myur5e_description)/urdf/mytable.urdf.xacro" />
> 
>   
>   <xacro:include filename="$(find myur5e_description)/urdf/my_overview_camera.urdf.xacro" />
>   
>   <link name="world" />
> 
>   <xacro:property name="table_height" value="0.95" />
> 
>   <joint name="world_joint" type="fixed">
>     <parent link="world" />
>     <child link = "mypedestal" />
>     <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0" />
>   </joint>
> 
>   <joint name="world_to_table" type="fixed">
>     <parent link="world"/>
>     <child link="mytable"/>
>     <origin xyz="0.8 0.0 ${table_height}" rpy="0 0 0"/>
>   </joint>
> 
>   <joint name="robot_to_pedestal" type="fixed">
>     <parent link="mypedestal" />
>     <child link = "base_link" />
>     <origin xyz="0.0 0.0 ${pedestal_height}" rpy="0 0 0" />
>   </joint>
> 
>   <joint name="table_to_overcamera" type="fixed">
>     <parent link="mytable"/>
>     <child link="overcamera_link"/>
>     <origin xyz="0.75 0.0 0.8" rpy="${pi} ${3*pi/4} 0"/>
>   </joint>
>   <xacro:include filename="$(find myur5e_description)/urdf/gzplugin_grasp_fix.urdf.xacro"/>
>   <xacro:gzplugin_grasp_fix/>
> </robot>
>

ZZWang21, 2022-11-23:
Sorry, I did according to the posts, but it still cannot grasp the object. I am using ROS MELODIC, in UBUNTU 18.04. I am using a UR5e to grasp a cube model, which is a urdf.xacro model. I have set the plugin, and set the physical parameter of the cube, but it still slip away from the gripper? 

[image]

Any expert could help? Thank you very much. 
@JenniferBuehler

GradyM2M, 2022-11-23:
@ZZWang21 Hi, I think this plugin should be used in gazebo. I think this plugin should be used in gazebo. You should allow the gripper to wait for a short time when it is about to grab the target object, this will ensure that the plugin works.

ZZWang21, 2022-11-23:
@GradyM2M  Thanks for your reply. I am using it on Gazebo, and I can confirm it is working since I received messages on the Terminal. But I am trying to use RVIZ to control the robot, and the object-cube, in my case is only in Gazebo, but not in RVIZ. I do not know if this is the problem? 

May I ask, how do you control the gripper？ Thanks.

Eser-TU, 2024-03-20:
@JenniferBuehler Hi JenniferBuehler, Does this plugin work for ROS 2? Or should I move my code to ROS 1? I am using Ros2 Humble and Ubuntu Linux 22.04. Do you have any suggestions? Thanks.

JenniferBuehler, 2024-03-20:
Hi @Eser-TU , this package is long due for a migration for ROS2, I have it on my schedule for the next few weeks. I wouldn't recommend to migrate packages to ROS1.
If you just want to use the Gazebo grasp plugin alone, that's not ROS dependent, you'd just have to adjust the cmake file to not use catkin. Shouldn't be a big deal.
If you need the ROS binding, I'd suggest to use the `ros1_bridge` for the time being, until the package is migrated.