JenniferBuehler/gazebo-pkgs #32: Handling Multiple Grasps?

mburge22, 2019-05-13:
Hi Jennifer, 
I am trying to simulate a hand-off between two grippers using your Gazebo Grasp plugin. I am trying to grip an object (cylinder) with a 2-dof gripper that is already being griped by another 2-dof gripper. I was hoping this would result in a simultaneous grasp between the object and the two grippers until the first gripper detaches from the object. Instead, the second gripper will not attach until the first gripper is detached. Is it possible to handle simultaneous grasps between grippers with this plugin? Or do you have any idea how I could achieve this?
Thanks,
Michael

JenniferBuehler, 2019-05-13:
Hi Michael,

this would require modifications in the plugin, so unfortunately this won't work at the moment :( I'm also not sure how it would behave - think that the object is attached to the robot via a fixed joint (like it was part of the robot for the duration of attachment). If two independent kinematic chains grasp it, they would be connected - I suspect the physics engines wouldn't like that, at least not while any forces are applied to the joints. The robot(s) could possibly "explode" ;)

It would probably be better to switch the attachment instantaneously (at the same update step detach from one robot and attach to the other). But again, this has to be implemented in the plugin.

mburge22, 2019-05-13:
Hi Jennifer,

Thanks for the fast response. Also, thanks for the plugin which has helped us quite a bit! That is sort of what I experienced and now that I've looked through the source code and based on your description that makes sense. I think I am going to try a work around where I split the object I'm grasping into two links with a fairly rigid joint in between. Then I will disable collisions after attaching and try to attach one end-effector to one link and the other to the other link.

JenniferBuehler, 2019-05-13:
Ok, let me know how the object splitting goes, though to be honest I think that could still cause issues, even small forces pulling from different ends at the same object can cause conflicts.
I would perhaps detect in the plugin when a gripper is trying to grip the same object, and at the same instant switch the attachment over to the other gripper. Then the "reverse" attachment (the other gripper grabbing the object again) would have to be blocked for a certain amount of time, until the other gripper doesn't have any more contacts with the object. Disabling the collisions for this handover time is a very good idea.
Let me know how this goes, and if you get it working, I would be happy to look at a PR from you :)

mburge22, 2019-05-13:
That is a good idea. I considered doing something like that however, there is an incentive to be able to achieve a simultaneous grasp for our use case.  I will let you know how it goes! If I am unable to get my object splitting idea to work then an instantaneous switch between end-effectors as you suggested is probably our best option. If I end up working out the instantaneous switch I will send a PR. Thanks agian!