JenniferBuehler/gazebo-pkgs #33: Cannot grasp with my created robotic hand

Deastan, 2019-06-24:
Hello,

First, Thank you very much for your work. 

I am working on ubuntu 14, ros indigo and gazebo 2.2.6. I am using the simulation of the Kuka robot proved by: https://github.com/IFL-CAMP/iiwa_stack and a robotic hand that I created to simulate the robotic hand from Righthand takktle2.

It is a dummy question.. because I think I forgot something somewhere.. So excuse me for this question.

I installed your package in my catkin_ws and everything is built and worked. 
I ran my robot and try to grasp the object but the object is not caught but the contacts are done.
Here is a picture what I did to have an idea of the issue.[image]
And this one, when it was trying to grasp.
[image]
I didn't take the picture of the right moment but there is a contact of both side of the object and I sat 170degrees for the force angle tolerance. The other setting are :
`<forces_angle_tolerance>170</forces_angle_tolerance>
      <update_rate>100</update_rate>
      <grip_count_threshold>2</grip_count_threshold>
      <max_grip_count>10</max_grip_count>
      <release_tolerance>0.05</release_tolerance>
      <disable_collisions_on_attach>true</disable_collisions_on_attach>
      <contact_topic>__default_topic__</contact_topic>`

My question is: do I have to set somewhere the name of the object ? here it is only a rectangle. Or what did I miss ?

Warm regards,

Jonathan

JenniferBuehler, 2019-06-27:
Hi Jonathan,

can you post the output on the command line please?
Also, try to visualize the collision shapes (I think it's in the "View" menu) and see if contacts happen with the collision shapes, or only with the visual shapes you are seeing on your screen.
You can also try to reduce the `forces_angle_tolerance` and increase the `grip_count_threshold`.

Best regards,
Jennifer

JenniferBuehler, 2019-06-27:
Sorry I didn't think to check the output on your screenshot. I can't see any collision output from the grasp plugin. Try to visualize the collision shapes and let me know if the object and the hand are actually colliding.

Deastan, 2019-07-01:
I have reduced the forces_angle_tolerance and I have also increased the grip_count_threshold. But nothing change. 

I tried as you recommand to check the collision, so here this two pictures.
This one seems that it doesn't have any contact with the object...
[image]
But in this one, it seems to have a contact. It is a weird behavior... Why one finger will be like a fantom and the 2 others work ? I checked the urdf but everything seem to be good.. 
[image]

So maybe the issue comes from my side.. :/

JenniferBuehler, 2019-07-07:
Yes on the two screenshots below I can see there are contacts with one side of the gripper, but I can't see if there are also contacts from the opposing finger (the side where it has only one finger?).. Can you send a screenshot where you can see both sides? It looks to me that the finger is kind of inside the object...

Deastan, 2019-07-08:
Hello, I have fixed the issue! It came from my reinforcement learning algorithms. 
So to everyone, this was the issue...: 
```
self.reset_simulation_proxy = rospy.ServiceProxy('/gazebo/reset_simulation', Empty)
        self.reset_world_proxy = rospy.ServiceProxy('/gazebo/reset_world', Empty)
```
Be careful when you reset the environment of gazebo.

And to answer at my own question, we don't need to specify the name of the object that we want to catch.
Thank you very much for trying to help me and good job with what you have done with this patch.

closed 2019-07-08 by Deastan

JenniferBuehler, 2019-07-13:
Glad you solved it, thanks for posting the solution!