JenniferBuehler/gazebo-pkgs #35: Correlate gazebo_grasp_plugin with Moveit!

orsalmon, 2019-08-11:
Hi,

What is the best way to correlate between gazebo_grasp_plugin attach/detach and moveit pick & place success/failure (i.e. attach/detach objects from the planning scene)?

Currently, I have situations where the pickup by moveit is failed but the object is attached by the gazebo plugin and vice versa. 

Thanks

JenniferBuehler, 2019-08-13:
Can you provide more detail about "pickup by moveit failed" please? In Gazebo, the object has been grasped successfully? How did it then fail in moveit?
Thanks for the details :)

corot, 2019-08-30:
Interesting point; I suppose what @orsalmon wants is that pickup by moveit fails if the grasp plugin doesn't attach any object, and so the pick action won't change the object status as attached.

I understand "pickup by moveit failed" as that the pickup action offered by MoveIt! fails, typically cause it cannot find a valid plan to reach the grasping pose. But afaik, it wont fail due to a "failed grasping".

All that said, I have no idea how to implement the connection, but I'll take a look, as it would be a very nice to have.

btw, @JenniferBuehler, great pluggin. Did you consider releasing it? Imho the code is worth the effort.

JenniferBuehler, 2019-09-08:
Sorry for the late response. 
Yes that would indeed be an interesting point about communicating the "failed" grasp. This could be done by publishing a message from Gazebo whenever an object has been attached or released. If no such "attach" message arrives, then the MoveIt attachment should not be done either.
I have put this in the features list to implement at some point, see #36. Hopefully I'll get around to it soon, though my time is pretty limited at the moment :(

Thanks @corot, I'm glad you like the plugin. Yes I did consider releasing it, at some point I did release a bunch of my packages, but I can't remember why I didn't do the gazebo-pkgs, there was some reason for it, which one that was slipped my mind... anyhow the main thing about releasing in general is that I found (with my other released packages) that people were anyway better off using the source in the end, to benefit from the newest updates. The releases weren't really used that much, at least not to my knowledge (maybe it also was because only people using the source opened issues).
So I didn't really have more releases on the radar at the moment. I will however consider your feedback and will look into releasing the gazebo-pkgs, or at least split off the grasp plugin (which is the most widely used) and release that at least.
Or did you mean by "release" just a tag tracking history on this repo? I've understood your comment  as release which is built as deb package.