JenniferBuehler/gazebo-pkgs #36: New feature: send messages from grasp plugin on attach/detach

JenniferBuehler, 2019-09-08:
As discussed in #35, it would be useful to communicate from the grasp plugin when an object was attached or released. This should ideally be done by message, so external packages can subscribe to that information.
However one thing about this to keep in mind is that this doesn't really simulate the robot. The robot would not communicate a failed grasp either, unless it has force sensors, but that would then have to be simulated by itself, probably from a separate gazebo plugin. In other cases, this kind of "grasp success measure" would have to be done by another module, e.g. a camera sensor which watches the arm and reports whether the object is held. Only a system like that would then simulate the real world.

Nevertheless, for testing and development purposes some kind of feedback from the plugin would be useful. Therefore it is not on the "features to be implemented" list.

corot, 2020-12-24:
:+1:  for this feature; as u explain, I use effort to confirm grasping with the real robot, but gazebo controller doesn't provide it, so I need something like this to mimic the behavior.

JenniferBuehler, 2021-03-26:
Thanks @corot for the PR, good work! I commented on it there. I will free up some time in my schedule to push this much-wanted feature in the next weeks. Some adjustments are required to keep the code ROS-independent.

closed 2021-07-15 by JenniferBuehler