JenniferBuehler/gazebo-pkgs #37: Can I spawn a opbject by existed model.sdf instead of a hard_code streamstring

TouchDeeper, 2019-10-01:
Hello

The cube_spawner spawn a cube by hard-code streamstring. Is there a way to use an existed model.sdf file.

matbol, 2020-07-22:
Try something like:

```

std::ifstream file("/path2/model.sdf"); 
gazebo_msgs::SpawnModel model;
	
if (file.is_open()) {
	std::string line;
	while (std::getline(file, line)) {
                   model.request.model_xml+=line;
	}
	file.close();
}

model.request.model_name="YourOwnModel";
model.request.initial_pose.position.x = 4;
model.request.initial_pose.position.y = 4;
model.request.initial_pose.position.z = 4;
client_spwn.call(model);
```

TouchDeeper, 2020-07-25:
@matbol I test your code, but there is no model occur in gazebo.
```
        std::ifstream file;
        file.open(path);
        if(!file){
            std::cout<<"can't open "<<path<<std::endl;
            exit(-1);
        }
        gazebo_msgs::SpawnModel model;
        ros::ServiceClient client_spwn = nh.serviceClient< gazebo_msgs::SpawnModel> ("/gazebo/spawn_urdf_model");
        if (file.is_open()) {
            std::string line;
            while (std::getline(file, line)) {
                model.request.model_xml+=line;
            }
            file.close();
        }

        model.request.model_name = model_name;
        model.request.initial_pose.position.x = 4;
        model.request.initial_pose.position.y = 4;
        model.request.initial_pose.position.z = 4;
        client_spwn.call(model);
```

The model file is [here](https://drive.google.com/drive/folders/1KUina5Nlplq1PURf2snCHbe4QQFsYWMb?usp=sharing).
Coud you please provide some advice? Thanks a lot!

matbol, 2020-07-27:
Instead
`ros::ServiceClient client_spwn = nh.serviceClient< gazebo_msgs::SpawnModel> ("/gazebo/spawn_urdf_model");`
use 
`ros::ServiceClient client_spwn = nh.serviceClient< gazebo_msgs::SpawnModel> ("/gazebo/spawn_sdf_model");`
for .sdf files.

JenniferBuehler, 2020-12-07:
Closing due to inactivity. Does anyone still need help with this? Then please re-open.

closed 2020-12-07 by JenniferBuehler