JenniferBuehler/gazebo-pkgs #38: Error: The Gazebo Grasp Plugin will not Work

Murtaza786-oss, 2019-10-22:
Hello I am using a UR5 with Robotiq85 Gripper to grasp an object. I was facing problems initially because the object would slip out of the fingers and fall, then I found your plugin which seems to be promising to fix this error. I have attached your plugin into the URDF which is,

```
  <gazebo>
   <plugin name="gazebo_grasp_fix" filename="libgazebo_grasp_fix.so">
       <arm>
          <arm_name>gazebo_gripper</arm_name>
          <gripper_link>robotiq_85_left_finger_tip_link</gripper_link>
          <gripper_link>robotiq_85_right_finger_tip_link</gripper_link>
          <palm_link>robotiq_85_base_link</palm_link>
       </arm>
       <forces_angle_tolerance>100</forces_angle_tolerance>
       <update_rate>4</update_rate>
       <grip_count_threshold>4</grip_count_threshold>
       <max_grip_count>8</max_grip_count>
       <release_tolerance>0.005</release_tolerance>
       <disable_collisions_on_attach>false</disable_collisions_on_attach>
       <contact_topic>__default_topic__</contact_topic>
    </plugin>
  </gazebo>
</robot>
```
The palm links and the gripper links are alright but when I run the code I get the following error 
```
**[ INFO] [1571737349.566807982, 0.711000000]: Loaded gazebo_ros_control.
[Msg] Loading grasp-fix plugin
[Msg] GazeboGraspFix: Using disable_collisions_on_attach 0
[Msg] GazeboGraspFix: Using update rate 4
[Msg] GazeboGraspFix: Using max_grip_count 8
[Msg] GazeboGraspFix: Using grip_count_threshold 4
[Msg] GazeboGraspFix: Using release_tolerance 0.005
[Err] [GazeboGraspGripper.cpp:64] GazeboGraspGripper: Palm link robotiq_85_base_link not found. The gazebo grasp plugin will not work.
[Err] [GazeboGraspFix.cpp:230] GazeboGraspFix: Could not initialize arm gazebo_gripper. Skipping.
[Err] [GazeboGraspFix.cpp:260] ERROR: GazeboGraspFix: Cannot use a GazeboGraspFix because no arms were configured successfully. Plugin will not work.**
```
I would be glad if you can help me

corot, 2019-10-23:
Just in case (as I found this problem myself), palm and gripper links are Gazebo model links, not ROS URDF model links.

Murtaza786-oss, 2019-10-23:
@corot thank you....but I am still not clear because normally the gripper links and palm links are taken from the robotiq85 urdf.....what should the palm links and gripper links be in this case?

corot, 2019-10-23:
Not here; check the names of the matching links on gazebo model (left panel of gazebo gui). Could possibly be the case that you must modify the SDF model to split links, as gazebo join all statically joined links.

JenniferBuehler, 2019-10-27:
Hello, sorry for my late response on this. Did you manage to fix it? As @corot suggested, you can look up the names on the Gazebo info pane on the left. They are taken from the urdf, but are prefixed with the the robot's namespace.

JenniferBuehler, 2020-12-07:
Closing because I haven't heard back if this has been resolved. Please re-open if you still need help with this.

closed 2020-12-07 by JenniferBuehler