JenniferBuehler/gazebo-pkgs #39: simulate gripper surface to bend when in contact

brinij, 2019-11-06:
Hi,

I have general question about Gazebo simulation capabilities.
Do you know if it is possible to simulate bending of, for example, shadow hand fingertips when they are in contact with some object? Should I change material of fingertip to be a rubber in SDF file or play with friction parameters? I want to achieve human-finger-like behavior.
Did you already achieved something like this in one of you simulations?
Thank you in advance!

JenniferBuehler, 2019-11-06:
Hi,

I think soft bodies are not suppoted yet, but I think bullet and dart engines support it to some extent, and if I remember correctly there is an effort to integrate this into Gazebo, though I don't know details and how much has bern done on that front. You'll probably get more informed answers on answers.gazebosim.org on the newest developments :)

In general I'd try to get thinks fully working with rigid bodies first, if it doesn't run smoothly with rigid then it will probably blow up with soft. Do you have a working setup with the rigid gripper yet?

brinij, 2019-11-07:
Thank you for the quick feedback.
Yes I have it working with rigid gripper in gazebo7.
Anyway thank you for your help.

JenniferBuehler, 2019-11-17:
Ok. I hope you found some answers online in regards to soft bodies. I'll close this for now, if you end up trying soft bodies through bullet or dart, feel free to re-open this to have a discussion about that here.

closed 2019-11-17 by JenniferBuehler