JenniferBuehler/gazebo-pkgs #41: Enable/disable plugin during script execution

jordan1203, 2020-05-28:
Hello !
At first, thanks a lot for your works, it helps me a lot !
I'm actually working on a gazebo project in order to grasp and push objects. I'm using your plugin for the gasp function but i got a problem: When i try to push an object, this object stay glue to the gripper because of this plugin. I would like to know if it's possible (or not) to disable a plugin when i'm executing the script and how to do it (maybe with a service...) ?

I'm working with an UR5 and a gripper85 and I got melodic version of Ubuntu.

Any help will be appreciated and thank in advance. Jordan

JenniferBuehler, 2020-05-28:
Hello!
I'm glad the packages help you :)
The obstacle should actually only get stuck when the gripper is "gripping" the obstacle from two sides... can you post a screenshot of how you are pushing the obstacle?

It would be easy to disable the plugin via service call, but such a service call hasn't been implemented for the plugin yet... however you could just extend it yourself, just make a flag (which is set true upon service call) and return in [OnUpdate](https://github.com/JenniferBuehler/gazebo-pkgs/blob/baf0f033475c3a592efb0862079f3ff8392cadf6/gazebo_grasp_plugin/src/GazeboGraspFix.cpp#L501) if it is set to true (maybe in the service call, detach all already attached objects).

I wouldn't permanently integrate any solution which uses ros services in the plugin, because it's meant to be ROS agnostic. A Gazebo service call / subscriber would be OK however. But then you'd have to wrap this gazebo service request into a ROS node which offers a ROS service, which is a bit more workaround. If you just want  a quick solution, I suggest you  just put a ROS service client into the plugin and set the flag in the service callback. Then you should have a working solution pretty quickly.

Hope this helps!

JenniferBuehler, 2020-09-20:
There has been no more discussion here, so I consider this closed. Please re-open if required.

closed 2020-09-20 by JenniferBuehler