JenniferBuehler/gazebo-pkgs #42: ERROR: the following packages/stacks could not have their rosdep keys resolved

MunimRayhan, 2020-07-07:
When I try to build **jaco-arm-pkgs** I get the following error after running 

_rosdep install -y --from-paths . --ignore-src --rosdistro kinetic_

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
jaco_gazebo: Cannot locate rosdep definition for [gazebo_joint_control]
jaco_joints: Cannot locate rosdep definition for [joint_trajectory_execution]
jaco_on_table: Cannot locate rosdep definition for [common_sensors]

Can you help me with this?

JenniferBuehler, 2020-07-18:
I haven't used it with rosdep before, possibly need to add rosdistro entries. Just compile and if there are errors relating to missing dependencies, install them. If you install the packages as described in the wiki install instructions, it should be fine though.

JenniferBuehler, 2020-12-07:
Sorry but I'm not getting around to adding rosdistro entries and then maintaining them, the jaco packages are only to be used for tutorials etc. So I'm afraid this won't be fixed. To use jaco, you will have to add the jaco packages manually to your catkin workspace. Maybe some time in the future when I have more time (I'm praying for this day ;)) I will get around to adding the jaco packages too.

closed 2020-12-07 by JenniferBuehler