JenniferBuehler/gazebo-pkgs #44: gzserver: symbol lookup error

SaailN, 2020-09-17:
I was searching for a robotic gripper plugin. Found this awesome repo i tested this plugin(gazebo_grasp_fix) in ROS kinetic it was working properly but when i tried to use melodic version of same plugin its giving following error and gazebo. 
[image]

JenniferBuehler, 2020-09-18:
Hi, thanks for reporting this. I will take a look over the weekend

JenniferBuehler, 2020-09-20:
Unfortunately I can't reproduce this right now, though I'm having another new issue on the new laptop on which I didn't run this package yet. This could prevent me from getting this. It may take me a few days to pin this other issue down (it's in a related package I use for testing), because I don't have a great deal of time to work on this.

I checked in the `gazebo_grasp_plugin/CMakeLists.txt` and I think I may have found the issue. Could you add `${catkin_LIBRARIES}` to the `target_link_libraries` as here:

```
 target_link_libraries(gazebo_grasp_fix                                          
   ${catkin_LIBRARIES}                                                           
   ${GAZEBO_LIBRARIES}                                                           
   ${Boost_LIBRARIES}                                                            
 ) 
```

and report if this solved the issue?

It's really about time I make melodic the standard and finish up support for it. Time flies, and now melodic is well into being the standard ;) I'll be working on this on the side throughout the week, so if you find any other issues, please report here :) Thanks!

SaailN, 2020-09-21:
Thanks @JenniferBuehler  for quick response, I tried the master branch of the package in ROS melodic, its working properly.

closed 2020-09-21 by SaailN