JenniferBuehler/gazebo-pkgs #45: Can't detect contact between gripper and piece

davidrmdt, 2020-10-13:
Hi @JenniferBuehler
I tried using your plugin for my setup of UR3 arm with a gripper I have designed myself. The problem is that in gazebo I can see the contacts when I visualize contacts in Gazebo (view -> contacts). But when I use "gz topic -e ~/robot/contacts" in the terminal, that doesn't shows anything. 
So the plugin is not receiving any info.
Please, could you help me whit that problem? 
Regards 
David

JenniferBuehler, 2020-10-19:
Hi David,

I will look at this as soon as I can, but I need a few more details. Ideally, the urdf of the robot, so I can test myself.
Are you using gazebo with ROS, or just by itself? Which version are you using?
Can you send the output of
`gz topic -l`
while gazebo is running with your robot loaded?

Also, are you sure the robot is colliding with sth (touching the obstacle)?You can try to make an obvious collision, e.g create a sphere and move it right next to the robot so it touches.

I presume you added collision shapes to the urdf? You can show them with the `view` menu as well. They should be displayed orange. Only contacts with collision shapes will be reported.

davidrmdt, 2020-10-19:
Hi @JenniferBuehler, thanks for answering
I upload my workspace so you can test. 
[tfm5_ws.tar.gz](https://github.com/JenniferBuehler/gazebo-pkgs/files/5403623/tfm5_ws.tar.gz)
I'm using gazebo and moveit with ros kinetik. 
In the src folder there is a image with the ouput of the terminal, and also you can see the contacts in gazebo.
In the urdf I have added collision shapes.
To run it I use: "roslaunch ur_gazebo ur3.launch" and in an other terminal "roslaunch ur3 demo_planning_execution.launch"

JenniferBuehler, 2020-10-23:
Hi, 
thanks for sharing the files. I didn't get around to fix the other issue which I have on my current setup yet, so haven't been able to test it, but I checked the code to refresh my memory, and the topic the contacts are published on is actually `~/physics/contacts`. Can you check that anything is output on that topic?

The other topic `~/<robot-name>/contacts` is a filtered topic. If this is not available, it means something has gone wrong with loading the grasp plugin.  I guess you set up the gazebo grasp plugin as [described here](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin).

Could you please post the entire output after you launch  `roslaunch ur_gazebo ur3.launch`?

JenniferBuehler, 2020-10-23:
One way to double check the plugins are loaded correctly is to set verbose output for gazebo. In the launch file which brings up gazebo, you can set the `verbose` flag to true, e.g.

```
   <include file="$(find gazebo_ros)/launch/empty_world.launch">                                                    
     <arg name="verbose" value="true" />
     ..
```

If the plugin library file is not found for some reason, unfortuately the gazebo is only printed if you set the verbose mode.

davidrmdt, 2020-10-26:
Hi, this is the output when I launch gazebo:

david@david:~/tfm5_2_ws$ roslaunch ur_gazebo ur3.launch 
... logging to /home/david/.ros/log/8b7b9094-1770-11eb-b06a-484520727c66/roslaunch-david-29123.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

inconsistent namespace redefinitions for xmlns:xacro:
 old: http://wiki.ros.org/xacro
 new: http://ros.org/wiki/xacro (/home/david/tfm5_2_ws/src/ur3/urdf/objects/table.xacro)
started roslaunch server http://david:41555/

SUMMARY
========

PARAMETERS
 * /arm_controller/action_monitor_rate: 10
 * /arm_controller/constraints/elbow_joint/goal: 0.1
 * /arm_controller/constraints/elbow_joint/trajectory: 0.1
 * /arm_controller/constraints/goal_time: 0.6
 * /arm_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /arm_controller/constraints/stopped_velocity_tolerance: 0.05
 * /arm_controller/constraints/wrist_1_joint/goal: 0.1
 * /arm_controller/constraints/wrist_1_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_2_joint/goal: 0.1
 * /arm_controller/constraints/wrist_2_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_3_joint/goal: 0.1
 * /arm_controller/constraints/wrist_3_joint/trajectory: 0.1
 * /arm_controller/joints: ['shoulder_pan_jo...
 * /arm_controller/state_publish_rate: 25
 * /arm_controller/stop_trajectory_duration: 0.5
 * /arm_controller/type: position_controll...
 * /hand_controller/action_monitor_rate: 10
 * /hand_controller/constraints/finger_joint1/goal: 0.1
 * /hand_controller/constraints/finger_joint1/trajectory: 0.1
 * /hand_controller/constraints/finger_joint2/goal: 0.1
 * /hand_controller/constraints/finger_joint2/trajectory: 0.1
 * /hand_controller/constraints/goal_time: 0.6
 * /hand_controller/constraints/stopped_velocity_tolerance: 0.05
 * /hand_controller/joints: ['finger_joint1',...
 * /hand_controller/state_publish_rate: 25
 * /hand_controller/stop_trajectory_duration: 0.5
 * /hand_controller/type: position_controll...
 * /joint_group_position_controller/joints: ['finger_joint1',...
 * /joint_group_position_controller/type: position_controll...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.16
 * /use_sim_time: True

NODES
  /
    arm_controller_spawner (controller_manager/controller_manager)
    fake_joint_calibration (rostopic/rostopic)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    hand_controller_spawner (controller_manager/controller_manager)
    joint_state_controller_spawner (controller_manager/controller_manager)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ros_control_controller_manager (controller_manager/controller_manager)
    spawn_gazebo_model (gazebo_ros/spawn_model)

[... 57 lines cut ...]
[ERROR] [1603705775.051311481, 0.284000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_1_joint
[ERROR] [1603705775.053205705, 0.284000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_2_joint
[ERROR] [1603705775.055218945, 0.284000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_3_joint
[ERROR] [1603705775.057495898, 0.284000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/finger_joint1
[ERROR] [1603705775.059771955, 0.284000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/finger_joint2
[ INFO] [1603705775.068452074, 0.284000000]: Loaded gazebo_ros_control.
[ WARN] [1603705775.068687999, 0.285000000]: The default_robot_hw_sim plugin is using the Joint::SetPosition method without preserving the link velocity.
[ WARN] [1603705775.068734731, 0.285000000]: As a result, gravity will not be simulated correctly for your model.
[ WARN] [1603705775.068765441, 0.285000000]: Please set gazebo_pid parameters, switch to the VelocityJointInterface or EffortJointInterface, or upgrade to Gazebo 9.
[ WARN] [1603705775.068794692, 0.285000000]: For details, see https://github.com/ros-simulation/gazebo_ros_pkgs/issues/612
Loaded joint_state_controller
Started ['joint_state_controller'] successfully
Loaded hand_controller
Loaded joint_group_position_controller
Loaded arm_controller
Started ['hand_controller'] successfully
Started ['arm_controller'] successfully
[hand_controller_spawner-9] process has finished cleanly
log file: /home/david/.ros/log/8b7b9094-1770-11eb-b06a-484520727c66/hand_controller_spawner-9*.log
[arm_controller_spawner-8] process has finished cleanly
log file: /home/david/.ros/log/8b7b9094-1770-11eb-b06a-484520727c66/arm_controller_spawner-8*.log
[joint_state_controller_spawner-7] process has finished cleanly
log file: /home/david/.ros/log/8b7b9094-1770-11eb-b06a-484520727c66/joint_state_controller_spawner-7*.log
[ros_control_controller_manager-10] process has finished cleanly
log file: /home/david/.ros/log/8b7b9094-1770-11eb-b06a-484520727c66/ros_control_controller_manager-10*.log

JenniferBuehler, 2020-11-02:
Hi, sorry for the delays in my responses, so busy and in the evenings I tend to forget...
This output suggests the plugin isn't being loaded at all. It could be you're not loading the right urdf, or the plugin library file is not found. Can you switch on gazebo verbose mode (flag in launch file as I mentioned above) and post the output? That will show errors if the library file isn't found then.

davidrmdt, 2020-11-05:
Hi Jen,
I have turn the verbose on, and see that, as you say gazebo was not loading the plugin correctly. The error is a little bit shameful, I forgot to put these packages in my folder. https://github.com/JenniferBuehler/gazebo-pkgs
So now it works well.
I also was making my own system to attach objects to the gripper, but my system fails a lot.  
Thank you very much for your help

JenniferBuehler, 2020-11-05:
Phew I'm very glad this could be resolved :)
Good luck with your project!
Feel free to reopen the issue or open a new one if you have further questions.

closed 2020-11-05 by JenniferBuehler