JenniferBuehler/gazebo-pkgs #46: Is there any way to disable collisions from ros?

davidrmdt, 2020-11-19:
Hi @JenniferBuehler 
I have been looking for a way to disable collisions or physics in gazebo in order to simulate drilling holes in a piece. I would like to be able to do this from python. But I haven't seen a solution.
Do you know if there is a way for do this?
Regards
David

JenniferBuehler, 2020-11-19:
Hi David,
You could always create models without collision shapes, e.g. a wooden plank without a collision shape (only visual shape). However this has to be a static object then, otherwise it will just fall through the ground. Also, other obstacles which you may put on it will fall through then. Unless you make a hole into the collision shape where the robot is expected to drill (if that's a fixed spot). If I remember correctly, simulated laser scanners will also only see the collision shapes, but of that I'm currently not sure, would have to look it up / try out.
A dirty trick would be to make the object part of the robot with disabled self collisions.

So it depends on your setup. What exactly are you trying to do?

davidrmdt, 2020-11-20:
Hi, thanks for answering so fast.
I have tried to use an STL with the hole for the collision object and it seems to work. But the grasping plugin now fails a lot and it doesn't attach the piece to the gripper (sometimes It works, but only a few). Can it be related to using a mesh?

JenniferBuehler, 2020-11-20:
Hm, it could be related, was it a cube/sphere before? Mesh mesh collisions are generally a bit trickier/bugger, but it shouldn't be sooo bad as to fail that often.
If it's a cube you're talking about, you could build the collision shape out of 4 cubes (leaving space in the middle) maybe? Would be worth a try at least to find out if that was the issue

davidrmdt, 2020-11-20:
The piece is a cube. I have been changing values in the plugin, hopping to improve the results, but I haven't reached anything.
Hmm, you give me an idea. Maybe creating a thin cover for the mesh it should work. I'm gonna try it

davidrmdt, 2020-11-20:
Well, making a sandwich works jajaj. It's a dirty trick, but I don't see any other solution. Here is the result
[image]
Thank you very much Jen

JenniferBuehler, 2020-11-20:
Awesome, glad it worked!
I'll close this, if you have other questions please feel free to open a new issue or reopen this.
Good luck with your project!

closed 2020-11-20 by JenniferBuehler