JenniferBuehler/gazebo-pkgs #47: Build on ROS noetic

robopassio, 2020-12-02:
I tried to build this package, specifically, **gazebo_version_helpers** on ROS noetic.
I got the following error

```
In file included from /home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/src/GazeboVersionHelpers.cpp:1:
/home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/include/gazebo_version_helpers/GazeboVersionHelpers.h:116:14: error: deduced class type ‘Box’ in function return type
  116 | gz_math::Box GetBoundingBox(const T &t)
      |              ^~~~~~~~~~~~~~
In file included from /usr/include/ignition/math6/ignition/math.hh:29,
                 from /usr/include/sdformat-9.2/sdf/Param.hh:33,
                 from /usr/include/sdformat-9.2/sdf/Element.hh:28,
                 from /usr/include/sdformat-9.2/sdf/Actor.hh:25,
                 from /usr/include/sdformat-9.2/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Battery.hh:25,
                 from /usr/include/gazebo-11/gazebo/common/common.hh:8,
                 from /usr/include/gazebo-11/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-11/gazebo/gazebo.hh:20,
                 from /home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/include/gazebo_version_helpers/GazeboVersionHelpers.h:4,
                 from /home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/src/GazeboVersionHelpers.cpp:1:
/usr/include/ignition/math6/ignition/math/Box.hh:42:11: note: ‘template<class Precision> class ignition::math::v6::Box’ declared here
   42 |     class Box
      |           ^~~
In file included from /home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/src/GazeboVersionHelpers.cpp:1:
/home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/include/gazebo_version_helpers/GazeboVersionHelpers.h:126:36: error: template placeholder type ‘const Box<...auto...>’ must be followed by a simple declarator-id
  126 | GzVector3 GetBoundingBoxDimensions(const gz_math::Box &box);
      |                                    ^~~~~
In file included from /usr/include/ignition/math6/ignition/math.hh:29,
                 from /usr/include/sdformat-9.2/sdf/Param.hh:33,
                 from /usr/include/sdformat-9.2/sdf/Element.hh:28,
                 from /usr/include/sdformat-9.2/sdf/Actor.hh:25,
                 from /usr/include/sdformat-9.2/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Battery.hh:25,
                 from /usr/include/gazebo-11/gazebo/common/common.hh:8,
                 from /usr/include/gazebo-11/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-11/gazebo/gazebo.hh:20,
                 from /home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/include/gazebo_version_helpers/GazeboVersionHelpers.h:4,
                 from /home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/src/GazeboVersionHelpers.cpp:1:
/usr/include/ignition/math6/ignition/math/Box.hh:42:11: note: ‘template<class Precision> class ignition::math::v6::Box’ declared here
   42 |     class Box
      |           ^~~
In file included from /home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_grasp_plugin/src/GazeboGraspGripper.cpp:10:
/home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/include/gazebo_version_helpers/GazeboVersionHelpers.h:116:14: error: deduced class type ‘Box’ in function return type
  116 | gz_math::Box GetBoundingBox(const T &t)
      |              ^~~~~~~~~~~~~~
In file included from /usr/include/ignition/math6/ignition/math.hh:29,
                 from /usr/include/sdformat-9.2/sdf/Param.hh:33,
                 from /usr/include/sdformat-9.2/sdf/Element.hh:28,
                 from /usr/include/sdformat-9.2/sdf/Actor.hh:25,
                 from /usr/include/sdformat-9.2/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Battery.hh:25,
                 from /usr/include/gazebo-11/gazebo/common/common.hh:8,
                 from /usr/include/gazebo-11/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-11/gazebo/gazebo.hh:20,
                 from /home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_grasp_plugin/src/GazeboGraspGripper.cpp:2:
/usr/include/ignition/math6/ignition/math/Box.hh:42:11: note: ‘template<class Precision> class ignition::math::v6::Box’ declared here
   42 |     class Box
      |           ^~~
In file included from /home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_grasp_plugin/src/GazeboGraspGripper.cpp:10:
/home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/include/gazebo_version_helpers/GazeboVersionHelpers.h:126:36: error: template placeholder type ‘const Box<...auto...>’ must be followed by a simple declarator-id
  126 | GzVector3 GetBoundingBoxDimensions(const gz_math::Box &box);
      |                                    ^~~~~
In file included from /usr/include/ignition/math6/ignition/math.hh:29,
                 from /usr/include/sdformat-9.2/sdf/Param.hh:33,
                 from /usr/include/sdformat-9.2/sdf/Element.hh:28,
                 from /usr/include/sdformat-9.2/sdf/Actor.hh:25,
                 from /usr/include/sdformat-9.2/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Battery.hh:25,
                 from /usr/include/gazebo-11/gazebo/common/common.hh:8,
                 from /usr/include/gazebo-11/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-11/gazebo/gazebo.hh:20,
[... 53 lines cut ...]
   42 |     class Box
      |           ^~~
/home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/src/GazeboVersionHelpers.cpp: In function ‘gazebo::GzVector3 gazebo::GetBoundingBoxDimensions(...)’:
/home/robo/catkin_ws/src/gazebo-grasp-pkgs/gazebo_version_helpers/src/GazeboVersionHelpers.cpp:309:18: error: ‘box’ was not declared in this scope
  309 |     GzVector3 bb(box.XLength(), box.YLength(), box.ZLength());
      |                  ^~~

```

The errors are summarized as follows:
```
error: deduced class type ‘Box’ in function return type
  116 | gz_math::Box GetBoundingBox(const T &t)
      |              ^~~~~~~~~~~~~~

error: template placeholder type ‘const Box<...auto...>’ must be followed by a simple declarator-id
  126 | GzVector3 GetBoundingBoxDimensions(const gz_math::Box &box);
      |                                    ^~~~~

 error: template placeholder type ‘const Box<...auto...>’ must be followed by a simple declarator-id
  126 | GzVector3 GetBoundingBoxDimensions(const gz_math::Box &box);
```
The reason seems from the c++17 on ROS noetic.
Could you please suggest to me a solution?
Thanks!

Sudhanshu-Dubey14, 2020-12-02:
Hey, I am facing the same issue on **ROS Melodic**. I am on **Arch Linux** and the Gazebo version is **11.3.0**.
I get the exact same error messages when I do `catkin_make`.

JenniferBuehler, 2020-12-02:
Hello,
I've been wanting to test on Noetic for a while now. Good incentive to do so. Will do in the next few days, and also double check with Melodic. Will let you know when it's done :)

JenniferBuehler, 2020-12-07:
There now is a `noetic` branch. Actually inappropriately named, should have just been called gazebo-11, but anyway... it will be merged soon ;)
Can you try if it works please?

Sudhanshu-Dubey14, 2020-12-07:
Hey @JenniferBuehler thanks a lot!!!
The packages have built on my system cleanly. I haven't tried using them yet but the error that I was getting while doing `catkin_make` is gone.
Thanks again.

JenniferBuehler, 2020-12-07:
I also tested it and it seems to work. So now it's merged into master :)
Will close this, please re-open if there is still an issue.

Thanks!

closed 2020-12-07 by JenniferBuehler