JenniferBuehler/gazebo-pkgs #48: Which topic I can subscribe to read the information indicating successful grasp?

zwj1996-xjt, 2020-12-22:
Thanks for your work. Now I want to read the state whether I am successful about the grasp. And I can see a line of info in green color on terminal, but I need to read the info by matlab (in another computer). I have tried topic /rosout_agg, but this topic would skip the green line, only focused on other information which is not belong to this gazebo-pkgs plug-in.

JenniferBuehler, 2020-12-24:
Hello,
No message is published upon successful grasp or detach, what you're seeing is a console print. I was thinking to add this feature, though it would then be a gazebo message, no rod message (the plugin is meant to stay rod independent). Not sure when I'll get around to add this feature.
But it's very easy to do. Where the plugin prints the console output, you could publish any message you want instead.

JenniferBuehler, 2020-12-24:
See also issue #36. This here is basically a duplicate, so if it's ok, I'll close this. Please re open if you consider this a separate issue :)

closed 2020-12-24 by JenniferBuehler