JenniferBuehler/gazebo-pkgs #49: Use gazebo_grasp_plugin but still cant grasp cube

Twilight89, 2021-02-02:
HI~ I run a grasp_demo on melodic gazebo9, using ur5 and robotiq_85_gripper. cube size is 0.05. I set the gripper close pose as 0.4, it just slipped between the two gripping. When set as 0.45, the two grippers shook, struggled to grasp the cube.
the parameters set as:   
			<forces_angle_tolerance>100</forces_angle_tolerance>
			<update_rate>20</update_rate>
			<grip_count_threshold>1</grip_count_threshold>
			<max_grip_count>3</max_grip_count>
			<release_tolerance>0.0198</release_tolerance> 
			<disable_collisions_on_attach>false</disable_collisions_on_attach>
			<contact_topic>__default_topic__</contact_topic>
Hope for your help, thanks~

JenniferBuehler, 2021-02-08:
Hi, do you get any output on the console from the grasp plugin? If it's loaded and the cube is attached, it shouldn't shake... Can you share your console output?

Twilight89, 2021-02-09:
Yep. I read the wiki and I know this plugin's principle is as long as there exists oppose force, then it can grasp the object.
When I launch the simulation environment, the console output about this plugin is:
[Msg] Loading grasp-fix plugin
[Msg] GazeboGraspFix: Using disable_collisions_on_attach 0
[Msg] GazeboGraspFix: Using update rate 100
[Msg] GazeboGraspFix: Using max_grip_count 30
[Msg] GazeboGraspFix: Using grip_count_threshold 5
[Msg] GazeboGraspFix: Using release_tolerance 0.005
[Msg] GazeboGraspFix: Adding collision scoped name robot::gripper_finger1_finger_tip_link::gripper_finger1_finger_tip_link_collision
[Msg] GazeboGraspFix: Adding collision scoped name robot::gripper_finger2_finger_tip_link::gripper_finger2_finger_tip_link_collision
[Msg] Subscribing contact manager to topic ~/robot/contacts
Loaded 'joint_state_controller'
[INFO] [1612833077.318443, 0.000000]: Calling service /gazebo/set_model_configuration
[INFO] [1612833077.327511, 0.000000]: Set model configuration status: SetModelConfiguration: success
(This part is green)
I'm so confused. Hoping for your help. Thanks!

plus:
there are also some errors about pid, and i searched, don't know whether it is associated. Outputs like follows:
[ERROR] [1612833077.091362132]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_pan_joint
[ERROR] [1612833077.092350657]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_lift_joint
[ERROR] [1612833077.093200189]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/elbow_joint
[ERROR] [1612833077.094025414]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_1_joint
[ERROR] [1612833077.094807872]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_2_joint
[ERROR] [1612833077.095554537]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_3_joint
[ WARN] [1612833077.095672752]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'gripper_finger1_joint'.
[ERROR] [1612833077.096404811]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/gripper_finger1_joint
[ INFO] [1612833077.104201384]: Loaded gazebo_ros_control.

JenniferBuehler, 2021-02-11:
Ok so the plugin is loaded, that's good. Do you get output when grasping happens too? There should be a message like "attaching object", and "detaching object" when it's detached again.
If you don't get this, the plugin never gets active. That could be due to the wobbling of the hand (forces never applied long enough).

I don't know your robot configuration, but the PID errors are from ros_control. It means there may be something not set up properly, but if you can control the robot well otherwise, and the problem only arises when grasping, then I suspect the controller config may not be the problem (it could be related, but not sure as I don't know your setup)

JenniferBuehler, 2021-02-11:
It could help if you display the contacts in Gazebo and send me a close-up screenshot/video of when it's grasping the object

Twilight89, 2021-02-11:
Thanks for your rely,  I am sorry that I cant upload the video and I thought just a screenshot maybe helpless.  
[Msg] GazeboGraspFix: Attaching cube2::link::collision to gripper UR5Arm.
[Msg] GazeboGraspFix: Detaching cube2::link::collision from gripper UR5Arm.

Twilight89, 2021-02-11:
Thanks for your rely, I put the gifs in my blog(in the [bottom).]([url](https://blog.csdn.net/a17381562089/article/details/113525809))
I found that when gripper can grasp(<release_tolerance>0.0050</release_tolerance>), the console outputs:
"[Msg] GazeboGraspFix: Attaching cube2:link:collision to gripper UR5Arm."
and when release object, outputs:
"[Msg] GazeboGraspFix: Detaching cube2:link:collision from gripper UR5Arm."
(But the gripper vibrated a little.)
when I set <release_tolerance>0.0010</release_tolerance>, outputs continuously:
"[Msg] GazeboGraspFix: Attaching cube2:link:collision to gripper UR5Arm."
"[Msg] GazeboGraspFix: Detaching cube2:link:collision from gripper UR5Arm."
the object just move a bit but not grasped.
So I just think that maybe the release_torleance is important? But why in the first <0.005> one, the gripper is vibrating? Is the normal?
Hope for rely, thanks~
And Happy Chinese New Year~

Twilight89, 2021-02-11:
Thanks for your rely, I put the gifs in my blog(in the bottom).
https://blog.csdn.net/a17381562089/article/details/113525809
I found that when gripper can grasp(<release_tolerance>0.0050</release_tolerance>), the console outputs:
"[Msg] GazeboGraspFix: Attaching cube2linkcollision to gripper UR5Arm."
and when release object, outputs:
"[Msg] GazeboGraspFix: Detaching cube2linkcollision from gripper UR5Arm."
(But the gripper vibrated a little.)
when I set <release_tolerance>0.0010</release_tolerance>, outputs continuously:
"[Msg] GazeboGraspFix: Attaching cube2linkcollision to gripper UR5Arm."
"[Msg] GazeboGraspFix: Detaching cube2linkcollision from gripper UR5Arm."
the object just move a bit but not grasped.
So I just think that maybe the release_torleance is important? But why in the first <0.005> one, the gripper is vibrating? Is the normal?
Hope for rely, thanks~
And Happy Chinese New Year~

JenniferBuehler, 2021-02-11:
The release tolerance must be higher, otherwise the cube will be detached easily.

So with the higher release tolerance of 0.005 the object stayed attached, but the arm vibrated while it was holding the cube?

JenniferBuehler, 2021-02-11:
What are the properties of your object? If it's very heavy or has a weird inertial then it will affect the robot model. The object basically becomes part of the robot while it's grasping the object. That could lead to strange behaviour with the physics if your object has properties which mean in reality the robot would not be able to grasp it (eg very heavy)

Twilight89, 2021-02-11:
Hi, as you can see in my blog's gifs, now the problem is with the higher release tolerance of 0.005, the object can be grasped, but the gripper vibrated while grasping the cube.
The cube's property is normal belows: (in the .world file), which I think is normal.
<inertial>
          <mass>0.0565</mass>
          <inertia>
            <ixx>2.3541666666666672e-05</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>2.3541666666666672e-05</iyy>
            <iyz>0</iyz>
            <izz>2.3541666666666672e-05</izz>
          </inertia>
        </inertial>
        <collision name="collision">
          <geometry>
            <box>
              <size>0.05 0.05 0.05</size>
            </box>
          </geometry>
          <surface>
            <contact>
              <poissons_ratio>0.347</poissons_ratio>
              <elastic_modulus>8.8e+09</elastic_modulus>
              <ode>
                <kp>100000</kp>
                <kd>1</kd>
                <max_vel>1.0</max_vel>
                <min_depth>0.001</min_depth>
              </ode>
            </contact>
            <friction>
              <torsional>
                <coefficient>10.0</coefficient>
                <use_patch_radius>0</use_patch_radius>
                <surface_radius>0.05</surface_radius>
              </torsional>
            </friction>
          </surface>
        </collision>

Plus:Can you click in my blog and see the gifs in the bottom?

Twilight89, 2021-02-11:
Hi, as you can see in my blog's gifs, now the problem is with the higher release tolerance of 0.005, the object can be grasped, but the gripper vibrated while grasping the cube.
The cube's important properties is normal belows: (in the .world file), which I think is normal.
        <inertial>
          <mass>0.0565</mass>
          <inertia>
            <ixx>2.3541666666666672e-05</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>2.3541666666666672e-05</iyy>
            <iyz>0</iyz>
            <izz>2.3541666666666672e-05</izz>
          </inertia>
        </inertial>
        <collision name="collision">
          <geometry>
            <box>
              <size>0.05 0.05 0.05</size>
            </box>
          </geometry>
          <surface>
            <contact>
              <poissons_ratio>0.347</poissons_ratio>
              <elastic_modulus>8.8e+09</elastic_modulus>
              <ode>
                <kp>100000</kp>
                <kd>1</kd>
                <max_vel>1.0</max_vel>
                <min_depth>0.001</min_depth>
              </ode>
            </contact>
            <friction>
              <torsional>
                <coefficient>10.0</coefficient>
                <use_patch_radius>0</use_patch_radius>
                <surface_radius>0.05</surface_radius>
              </torsional>
            </friction>
          </surface>
        </collision>
Plus:Can you click in my blog and see the gifs in the bottom?

JenniferBuehler, 2021-02-12:
I just scrolled through the blog and I can see the gif. I can't see much vibration in the gif, maybe only very minimal. Is this kind of vibration problematic for your setup, or were you only wondering where that came from?
 
The grasp plugin worked though. The vibration wouldn't come from the plugin directly, unless you don't see a fluctuating "attaching... detaching... attaching... etc" output. The object is fixed to the palm while it's attached/grasped, and becomes part of the robot. So the vibration could come if the attachment of the cube is a problematic "addition" to the robot for the physics engine.

Maybe try a different link as `<palm_link>`, just to test if that has an effect?

Twilight89, 2021-02-13:
Thanks, so the plugin is working, maybe the problem came from the gripper's(robotiq_85) palm_link. I'll try later. 
I found that sometimes the parameter set in gzplugin would effect the level of vibration of the gripper. In the gif, as you saw, maybe only very minimal.
Thanks again~

JenniferBuehler, 2021-02-14:
No problem. Let me know if you find out what caused the vibration. It's possible that it comes from the grippers applying pressure to the cube (which then basically becomes part of the palm).
I'll close this in the meantime. Please re-open if needed :)

closed 2021-02-14 by JenniferBuehler