JenniferBuehler/gazebo-pkgs #51: Not able to attach a custom object to kinova arm using grasp plugin

JBVAkshaya, 2021-04-16:
@JenniferBuehler I am trying to attach a door handle to Kinova arm (j2s7s300) using the grasp fix plugin but attaching the object is failing.
The plugin is getting loaded and contacts between palm-link, gripper fingers, and door are getting detected. But the end-effector is not getting attached to the door handle.

Below is a snippet of log info:
........................................................................................................................................................................................................
[ INFO] [1618533756.587926328, 57.770000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1618533757.192667533, 58.375000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533757.192698646, 58.375000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1618533757.193214415, 58.375000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533757.193233499, 58.375000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1618533757.193966685, 58.376000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533757.193986511, 58.376000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1618533757.194518915, 58.376000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533757.194535909, 58.376000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ERROR] [1618533757.794550317, 58.976000000]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1618533757.794582298, 58.976000000]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1618533757.794596924, 58.976000000]: No solution found after 1.206862 seconds
[ INFO] [1618533757.794614157, 58.976000000]: Unable to solve the planning problem
[ INFO] [1618533757.794640447, 58.976000000]: Planning attempt 4 of at most 5
[ WARN] [1618533757.795318745, 58.977000000]: Cannot find planning configuration for group 'arm' using planner 'RRTConnectkConfigDefault'. Will use defaults instead.
[ INFO] [1618533757.795456259, 58.977000000]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1618533757.795690701, 58.977000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1618533758.398923871, 59.580000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533758.398951891, 59.580000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1618533758.399524157, 59.581000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533758.399541047, 59.581000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1618533758.400087164, 59.581000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533758.400101906, 59.581000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1618533758.400516263, 59.582000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533758.400534927, 59.582000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ERROR] [1618533758.992230005, 60.173000000]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1618533758.992263935, 60.173000000]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1618533758.992283547, 60.173000000]: No solution found after 1.196777 seconds
[ INFO] [1618533758.992305717, 60.173000000]: Unable to solve the planning problem
[ INFO] [1618533758.992340537, 60.173000000]: Planning attempt 5 of at most 5
[ WARN] [1618533758.992964886, 60.174000000]: Cannot find planning configuration for group 'arm' using planner 'RRTConnectkConfigDefault'. Will use defaults instead.
[ INFO] [1618533758.993120151, 60.174000000]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1618533758.993372999, 60.174000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1618533759.609864166, 60.790000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533759.609909686, 60.791000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1618533759.610900039, 60.791000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533759.610922670, 60.791000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1618533759.611233089, 60.792000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533759.611251689, 60.792000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1618533759.611589541, 60.792000000]: Found a contact between 'door_1' (type 'Robot link') and 'j2s7s300_link_7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1618533759.611608439, 60.792000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
..............................................................................................................................................................................................................

I am not able to understand why is the door handle not getting attached to the gripper. For reference, I have uploaded the xacro files at https://github.com/JBVAkshaya/grasp_fix_plugin_custom_object

It will be of great help if you can give me some pointers on how can I fix this.

JenniferBuehler, 2021-04-17:
Hello,
I can actually not see any output from the plugin. The output about collisions comes from some ROS node loaded (possibly a row gazebo node). Can you post the whole output, or append as file, please?

JenniferBuehler, 2021-04-17:
Also see [this](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/Troubleshooting) to ensure the plugin really is loaded

JBVAkshaya, 2021-04-19:
@JenniferBuehler Yes, the plugin is getting loaded in my setup. 

[image]

I have added the object as a part of j2s7s300_standalone.xacro. Hence, the object is recognized as a link rather than an object. I will try spawning the object separately and update it here. For reference, below is a screenshot of the model structure in my gazebo world. 

[image]

Thanks once again @JenniferBuehler

JenniferBuehler, 2021-04-19:
Yes if the object ist part of the robot in the add/urdf then it will be a self-collision. You need to spawn it separately. Let me know how it goes

JBVAkshaya, 2021-04-20:
@JenniferBuehler Yes. It is working after spawning the object separately. Thank you for helping me through this.

closed 2021-04-20 by JBVAkshaya