JenniferBuehler/gazebo-pkgs #52: Can't attach object using robotiq 140 gripper

keke-220, 2021-04-17:
Hi,

I'm trying to use ur5 with robotiq 140 gripper to grasp object.
The plugin seems loaded but when the fingers are making contacts with the object, there is no "attached" info shows in the console. 
[image]
Here is how I set up the parameters:
[image]

keke-220, 2021-04-17:
Also, I use gz topic -e /gazebo/default/robot/contacts comand to check, I can see that there are collisions of object with left and right inner pads at the same time.

keke-220, 2021-04-17:
Oh, I just decreased grip count threshold to 4 and the object can be attached to gripper. However, it still takes like 20-30 seconds to see the "attached" info.

JenniferBuehler, 2021-04-17:
Huh, never heard about that kind of tremendous delay with the console output...  
Otherwise it works now? Detachment too?

keke-220, 2021-04-17:
Thank you for your reply!

I decrease the force_angle_tolerance to 90 and it works!
However, new issue comes  in which is when I open the gripper, the object is still sticking on one of the fingers for 1-2 second. I want the object to fall off immediately when the gripper is open. How to solve this issue?

JenniferBuehler, 2021-04-17:
You could try to decrease the max_grip_count and/or decrease the release_tolerance. See also description in [the wiki](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin).
Let me know if it works ☺️

khs-sm, 2021-09-25:
Hi, I'm also experiencing the same issue where I couldn't grasp my object with the same gripper "robotiq 140" as OP. I have changed the parameters in the plugin exactly from the previous post but the object still slipped every time. When I checked with the log, there is no errors showing.

Initially, I thought maybe the gripper problem so I tried with another URDF file and followed the steps from this [tutorial](https://www.youtube.com/watch?v=XENoj1lV8gU). Unfortunately, the object slipped again and I'm out of ideas on this. Any suggestion will be appreciated. Thank you.

Here I attached some pictures for the references. Video on [here](https://drive.google.com/file/d/1NXC6KyF-qd1JA4iFH_jfb_f0pMep4KVW/view?usp=sharing).

[image]
[image]
[image]

OS: Ubuntu 18.04 & ROS Melodic
Original Repo: [ur5_with_robotiq_gripper](https://github.com/intuitivecomputing/ur5_with_robotiq_gripper)

tattrongvu, 2022-04-06:
> Hi, I'm also experiencing the same issue where I couldn't grasp my object with the same gripper "robotiq 140" as OP. I have changed the parameters in the plugin exactly from the previous post but [...]

Did you find out how it work?