JenniferBuehler/gazebo-pkgs #53: Plugin is not loading

TomasMerva, 2021-04-18:
Hi, first of all I would like to thank you for this plugin. However, I am having a problem with loading it. Even though I applied it within my urdf file, after launching gazebo there is nothing mentioned about loading the plugin, as well as nothing is mentioned in gazebo. Do I need to somehow export Gazebo environment variables or something? Could you point me to the right direction? I believe that the problem is with the wrong location/path of plugin.

I have also tried things like these:
```
  <plugin name="gazebo_grasp_fix" filename="/home/tomas/catkin_ws/devel/lib/libgazebo_grasp_fix.so">
        ...
  </plugin>
```
Also do you have any idea why I am not getting any error and gazebo is just ignoring that plugin within urdf?

JenniferBuehler, 2021-04-18:
Hello,

As for all the gazebo plugins, you should add it to $GAZEBO_PLUGIN_PATH. The path should point to the folder where the .so file is located. Is your directory already in the plugin path?

JenniferBuehler, 2021-04-18:
You can also enable the output to see errors, see [this page](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/Troubleshooting).

TomasMerva, 2021-04-19:
Thank you for your fast reply. The `verbose arg` works fine but it still does not say anything about loading/ not loading grasp_plugin. I tried to add  $GAZEBO_PLUGIN_PATH according to [this page](https://answers.gazebosim.org//question/13391/how-to-set-gazebo_plugin_path-correctly-and-add-the-plugin-into-gazebo_ros/). 

1) I located the path to `libgazebo_grasp_fix.so` using 
```
locate 
/home/tomas/catkin_ws/devel/lib/libgazebo_grasp_fix.so
```
2) I checked the $GAZEBO_PLUGIN_PATH
```
export GAZEBO_PLUGIN_PATH=$PWD:$GAZEBO_PLUGIN_PATH
/home/tomas/catkin_ws/devel/lib:
```
But still I dont have any message in terminal about loading the plugin.

JenniferBuehler, 2021-04-19:
Hm strange, you should at least see some gazebo output though? Can you post the entire output?
If it's not displaying an error with the verbose flag, and neither plugin output is printed, one thing that would explain it is that the part of the urdf where you load the plugin isn't actually loaded. Do you define it in the urdf where all the rest of the model is defined? Or is it included from a separate file?

TomasMerva, 2021-04-20:
I am so sorry. I thought the same thing and finally Ive realized that I was working with the saved gazebo config and not with current urdf file... Your plugin works like charm, thank you very much for it and for your help. I am closing this issue

closed 2021-04-20 by TomasMerva