JenniferBuehler/gazebo-pkgs #54: GazeboGraspGripper: Palm link ee_link not found. The gazebo grasp plugin will not work

getupcurry, 2021-07-26:
Hi JenniferBuehler,

Thanks for creating GazeboGrasp plugin. I'm trying to use it with UR10 arm and RobotIq 85 gripper. But, the object falls as before and I get the following error:
[image]

I am using ROS Melodic + gazebo 9.  And the version of `gazebo_grasp_plugin` is master, because the `Melodic version` has some problems while running, the problem as follows:
```
gzserver: symbol lookup error: /home/ros/catkin_ws/devel/lib/libgazebo_grasp_fix.so: undefined symbol: _ZN6gazebo10GetPhysicsERKN5boost10shared_ptrINS_7physics5WorldEEE
Traceback (most recent call last)
```
Can you tell me whether the error in the picture is caused by the `gazebo_grasp_plugin` version?  And how to solve it?

getupcurry, 2021-07-26:
Here is how I set up the parameters:
	<gazebo>
		<plugin name="gazebo_grasp_fix" filename="libgazebo_grasp_fix.so">

            <arm>
                <arm_name>UR5Arm</arm_name>
                    <palm_link> gripper_finger1_inner_knuckle_link </palm_link>
		    <gripper_link> gripper_finger1_finger_tip_link </gripper_link>
		    <palm_link> gripper_finger2_inner_knuckle_link </palm_link>
		    <gripper_link> gripper_finger2_finger_tip_link </gripper_link>

            </arm>
			<forces_angle_tolerance>100</forces_angle_tolerance>
			<update_rate>20</update_rate>
			<grip_count_threshold>1</grip_count_threshold>
			<max_grip_count>3</max_grip_count>
			<release_tolerance>0.0198</release_tolerance> 
			<disable_collisions_on_attach>false</disable_collisions_on_attach>
			<contact_topic>__default_topic__</contact_topic>
		</plugin>
	</gazebo>

JenniferBuehler, 2021-07-26:
Hello,
The master branch is correct, melodic is old. I should delete it ;)
It seems to me that it is not finding your settings. Did you double check the inclusion in your urdf file?

getupcurry, 2021-07-26:
Thank you very much. I have solved the problem.

JenniferBuehler, 2021-07-26:
Ok, great. Will close this then :)

closed 2021-07-26 by JenniferBuehler