JenniferBuehler/gazebo-pkgs #57: ROS2 version for gazebo_grasp_plugin

AlezzzRd, 2022-05-22:
Hi Jennifer, 

Do you have a plan to develop a ros2 version for this very useful gazebo_grasp_plugin?

Thanks

JenniferBuehler, 2022-05-23:
Hi,
Yes I would like to migrate all my packages to ros2, however I don't know how soon I will be able to make time for it.

The grasp plugin itself is ROS independent though, it only depends on gazebo. You will only need to tweak the cmake file which uses catkin to get rid of ROS dependencies.

closed 2022-05-23 by JenniferBuehler

kongoncharuk, 2023-06-06:
Hi @JenniferBuehler . I've migrated gazebo_grasp_plugin and gazebo_version_helpers (as grasp_plugin was depending on it) to ROS2 humble here: https://github.com/kongoncharuk/gazebo-pkgs/tree/humble. 

Could you perhaps create a "humble" branch in your repository so that I can create a pull request into it?

JenniferBuehler, 2023-06-10:
Awesome thanks for doing that! I will create the branch

JenniferBuehler, 2023-06-10:
I branched it off master, but thinking probably it should be noetic. What did you base yours on?