JenniferBuehler/gazebo-pkgs #58: Can't attach object using dh_robotics_ag95_gripper.Object slips away when grasping.

S1aoXuan, 2023-05-17:
Hi, I'm trying to use ur10e with dh_robotics_ag95_gripper to grasp a cube in gazebo.
My platform is Ubuntu 18.04, ROS Melodic and Gazebo9.
The plugin should be loaded.
[image]
But without any attach message at the time of grabbing, the cube will just be bounced away.
Here is how I set up the parameters:
[image]
I use gz topic -l and I can see topic /gazebo/ur5_cubes/robot/contacts, where ur5_cube is my gazebo world's name. But I can get nothing when using gz topic -e ~/robot/contacts. 
The process of grabbing the box is shown in the gif.
[image]

JenniferBuehler, 2023-05-24:
Hi,
sorry for the late response. When the object bounces away, the attachment has probably not happened. Do you see any output (messages sent) on `/gazebo/ur5_cubes/robot/contact` when the robot grasps the object? The topic name should be the same (`/gazebo/ur5_cubes/` is the prefix for "home" aka `~/` so it should be the same)

S1aoXuan, 2023-05-24:
Thanks for your reply.
When the robot grasped the object, it just bounced away and the terminal did not show attached or any other messages.

JenniferBuehler, 2023-05-24:
Did you subscribe with gz topic -e?

S1aoXuan, 2023-05-24:
When I use gz topic -i /gazebo/ur5_cubes/robot/contact, I get this
[image]
and run gz topic -e /gazebo/ur5_cubes/robot/contact only get timestamp
[image]

JenniferBuehler, 2023-05-24:
Then it means there are no collisions, so the grasp plugin cannot work. Not sure why the object bounces away, hard to tell without being able to debug. Are you doing any other object manipulation? Does it bounce away just when the gripper closes?

S1aoXuan, 2023-05-25:
Sorry I didn't reply in time yesterday. There are always a lot of different problems with programs :) 
I am just doing the grasp cubes simulation now, the gazebo environment is  a few cubes on the table, currently these cubes have this problem. If there is any progress afterwards I will also be the first to post here.
Thank you!

liujun0808, 2023-12-19:
> Sorry I didn't reply in time yesterday. There are always a lot of different problems with programs :) I am just doing the grasp cubes simulation now, the gazebo environment is a few cubes on the [...]

I think I've encountered the same problem as you, have you solved it?

JenniferBuehler, 2023-12-19:
Your cubes are otherwise colliding properly, e.g. with tables? They are not set to "static"? And while the robot grasps them, you get nothing out of the gazebo contacs topic with `gz topic -e` (see comment above)?

liujun0808, 2023-12-20:
> Your cubes are otherwise colliding properly, e.g. with tables? They are not set to "static"? And while the robot grasps them, you get nothing out of the gazebo contacs topic with `gz topic -e` (see comment above)?

[image]
I used these topic commands, why is nothing displayed?What should the correct situation look like?
Why can some people use this plug-in to complete it successfully? I'm going to collapse : (

JenniferBuehler, 2023-12-20:
That is when the robot grasps the cube?
The cube collision works otherwise, so collision shape set up correctly (see my questions above)?