JenniferBuehler/gazebo-pkgs #62: The problem of objects penetrating each other after the gripper clamps the object

Liyiyang66, 2023-10-20:
Hello, I want to collect the torque information generated by the collision at the joint, but after the gripper clamps the object, the object will pass through other objects. What should I do?
[image]
[image]

This is information about the object：
[image]
[image]

JenniferBuehler, 2023-10-24:
Hi, did you try to add `<self_collide>1</self_collide>` to the link of the object?

Liyiyang66, 2023-10-25:
oh，thanks for your advice, the problem has been solved.

JenniferBuehler, 2023-10-25:
Great, that's good to hear! May I asked what resolved it?

Liyiyang66, 2023-10-25:
I used gazebo to re-export the more detailed parameter configuration， and set the <self_collide>1</self_collide>, then the object will colide the table, the force and torque change as well.

closed 2023-11-21 by JenniferBuehler