JenniferBuehler/gazebo-pkgs #63: <disable_collisions_on_attach>true</disable_collisions_on_attach>, Where does changing this parameter affect the <self_collide>

Liyiyang66, 2023-11-22:
i set like this,<disable_collisions_on_attach>true</disable_collisions_on_attach>, to eliminate the jitter between the gripper and the object, but  after the object will cross the table. After successfully gripping an object, is there any way to eliminate the collision between the gripper and the object without affecting the collision between the object and the table?
[image]

JenniferBuehler, 2023-11-22:
Disabling self collisions, see #62, was not enough to prevent the wobbling?

Liyiyang66, 2023-11-23:
oh, sorry, there is still some small wobbling of finger_tip_link after gripper attached the object. So, i set the true of the <disable_collisions_on_attach> to prevent the wobbling, then there is no  wobbling of finger_tip_link.
However,  perhaps because this reason, the self-collision of objects is eliminatedt, and the collision between the object and the table cannot be detected.
i want to eliminate the collision between the gripper and the object after the gripper successfully grasps the object，by this way, i can prevent the small wobbling of finger_tip_link. But i wish to preserve the collision of the object and the table, because i want to collect the force data.

JenniferBuehler, 2023-11-23:
That is two separate things. `<disable_collision_on_attach>` completely disables the collisions of that link, while the `<self_collide>` flag prevents links of the same model to collide with each other (but they can still collide with other objects which are not part of the same model).

Ideally, the self collision should be disabled (set to 0) so that once the link becomes part of the model (on attach), there are no strange wobble effects. The `<disable_collisions_on_attach>` should be false, otherwise the object will not collide with the table.

Liyiyang66, 2023-11-23:
oh, i think i got it, thanks for your advice. i will have a try.