JenniferBuehler/gazebo-pkgs #64: Gripper penetrates object in Gazebo，why？

liujun0808, 2023-12-18:
When the clamping jaw is closed, the terminal prompts that the object has been grasped, but the clamping jaw remains closed until it penetrates the object. Why is this?
[image]
The object is defined by myself, the following is its description.
  <link
    name="center_01">
    <inertial>
      <origin
        xyz="0.05 0.05 -0.015665"
        rpy="0 0 0" />
      <mass
        value="0.97831" />
      <inertia
        ixx="0.00073743"
        ixy="5.9851E-13"
        ixz="-1.6345E-14"
        iyy="0.00073743"
        iyz="-2.1095E-15"
        izz="0.0012655" />
    </inertial>
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
    <geometry>
        <mesh
          filename="package://ur_description/meshes/object/center_01.STL" />
      </geometry>
      <material
        name="plain_carbon_steel">
        <color
          rgba="0.52941 0.54902 0.54902 1" />
      </material>
    </visual>
    <collision>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://ur_description/meshes/object/center_01.STL" />
      </geometry>
    </collision>
  </link>

  <gazebo reference="center_01">
    <material name="carbon_steel"
      script="model:/home/liujun/catkin_ws/src/ur_description/meshes/object/carbon_steel.material" />
    <kp>1000000.0</kp>
    <kd>1.0</kd>
    <mu1>1.16</mu1>
    <mu2>1.16</mu2>
    <maxVel>0</maxVel>
    <minDepth>0.001</minDepth>
    
  </gazebo>
</robot>
Has anyone encountered this situation? Looking forward to your reply, thank you!

liujun0808, 2023-12-18:
The object is defined by myself, the following is its description.
[image]
[image]

JenniferBuehler, 2023-12-18:
The object will be attached to the robot but if your control is still closing the gripper, it will keep moving. Though there won't be collisions causing the object and robot to fly off into space ;) the grasp plugin does not handle your grippe control, only the attachment. You should close your gripper only as much as you know is necessary to cause a collision with the object. If it goes a bit beyond, that doesn't matter. The object will still be released once you open the gripper again.

JenniferBuehler, 2023-12-18:
If you need a notification on when attachment happens, so you can stop the movement: an event message is published. See [this example](https://github.com/JenniferBuehler/gazebo-pkgs/blob/master/gazebo_grasp_plugin_ros/src/grasp_event_republisher.cpp) which is subscribing to it (if you are using ROS, you can also use the ROS message which is published by this code in the link)

liujun0808, 2023-12-19:
> If you need a notification on when attachment happens, so you can stop the movement: an event message is published. See [this [...]

Thank you!!!!!Your answer gave me new ideas. I will try it and report back to you if I get good results.

JenniferBuehler, 2023-12-19:
Glad I could help :)
I'll close this now, please reopen if there are more comments.

closed 2023-12-19 by JenniferBuehler