JenniferBuehler/gazebo-pkgs #65: gazebo_version_helpers failed to build

cmakelabs, 2024-01-07:
Hi
I am trying to build the packages using `catkin build`, I am using `ros-noetic` on ubuntu 22.04, with `gazebo-classic`, however the build fails due to issue in `gazebo_version_helpers`. Could you please help?
[image]

JenniferBuehler, 2024-01-14:
Hi,

The last version tested was noetic on Ubuntu 20.04. I have it on my schedule to remove gazebo_version_helpers some time in the next couple of months. In case you end up doing it before, it would be nice if you could post a PR. It's basically just removing the #ifdef statements, replace the function calls by the gazebo ones, and drop support for old gazebo versions.
Or just fixing the compiler issue at hand.

Sorry I can't be of more help right now, I currently lack the time to update these packages. But it's on my horizon.

cmakelabs, 2024-02-17:
It seems that we both lack the time for this simple task :)

JenniferBuehler, 2024-02-17:
Yes I'm sorry, it's been so busy here. It is moving up my to-do list though, I'm working my way through to get around to looking after my packages (long overdue). I'm planning to migrate all ROS packages to ROS2 within the next 1-2 months. Fingers crossed this timeline works out ;)