JenniferBuehler/gazebo-pkgs #68: Update on humble branch

yulat214, 2024-01-30:
Hi,

I have checked #57, but this does not work as is.
I want to be able to use the humble branch easily.
Can I then send a pull request instead of #57?

Thanks.

JenniferBuehler, 2024-02-07:
Hi, did you check https://github.com/kongoncharuk/gazebo-pkgs/tree/humble?
There was never a PR into the humble branch that resulted from #57. Sure, if you do get it to work, a PR would be welcome.

Otherwise I expect to get it done myself within the next month. I can let you know once it's done as soon as I'm done, if you didn't submit a PR until then

yulat214, 2024-02-08:
Thank you for your reply !

I checked https://github.com/kongoncharuk/gazebo-pkgs/tree/humble, but this does not work as is.
I have already fixed the plugin about grasping, so I will check the other parts.

What I am a little concerned about is that I can't find the ROS2 version of the package that is listed as a dependency. 
I will check a little more.
If it goes well, I will send you a PR. If you can do it, I would be glad to know.

Thanks.

JenniferBuehler, 2024-02-08:
Which dependency do you mean?

yulat214, 2024-02-09:
Dependency on the page https://github.com/JenniferBuehler/gazebo-pkgs/wiki/Installation.
Humble could not install it from apt.

It did not affect me in grasping, but I have not built the rest of the system and do not yet know the impact.

JenniferBuehler, 2024-02-09:
That's the install instructions, so I'm not quite sure what you mean. There are several package dependencies listed. Some are my packages, they have to be ported too.
I assume you are after the gazebo grasp plugin? In that case, you could try to manually extract just this directory. But if you wait about maybe 2-3 weeks there will be a humble version of just the gazebo grasp plugin.

yulat214, 2024-02-09:
I have already been able to migrate the gazebo grasp plugin to Humble.

I was hoping to be able to migrate other features as well.
In doing so, I was wondering if these dependencies are needed...
I may have misunderstood something.

If only grasping is ok, I will create a PR.

yulat214, 2024-02-09:
I will send you a PR for the grasping function only.
Sorry for asking so many questions, I am a beginner.

When I do a colcon build, I get a build error due to a package that is not related to grasping.
If I select a package and build it, I can use the grasp function, but how should I announce it?

JenniferBuehler, 2024-02-09:
No worries.

Don't worry about the PR for grasping only. That will be moved into its own repo some time this month anyway. 

Maybe you can patch things up on your side without worrying too much about things being clean, and once I've got things ported I'll let you know and you can just switch over?

JenniferBuehler, 2024-02-09:
> When I do a colcon build, I get a build error due to a package that is not related to grasping. If I select a package and build it, I can use the grasp function, but how should I announce it?

I'm not sure what you mean, and it's hard to tell without knowing which error.

emusman-lab, 2024-10-16:
@yulat214  @JenniferBuehler
Any updates on Humble Branch I have checked https://github.com/kongoncharuk/gazebo-pkgs/tree/humble but it gives compilation error in Humble as some packages are still in ROS1 format.

JenniferBuehler, 2024-10-16:
Sorry, there are still no updates on the humble branch on this repo. It was only created so the PR could be made onto it. I think the work on the branch you linked must have been stopped.

I will delete my branch because it causes confusion, understandably.

Moving this to ROS2 has been on my TODO list for so long, I just don't have the time to get around to it :( It's on my radar for way too long. I think I need to take a holiday so I can finally do it :laughing: :cry: